The-OpenROAD-Project/OpenSTA #1: Splash TCL causes Tcl error and immediately halts the program

RTimothyEdwards, 2018-10-04:
I found that attempting to run OpenSTA crashed immediately on a TCL error.  By printing out the decoded string passed to Tcl_Eval(), I discovered that tcl/Splash.tcl procedure "show_warranty" has non-ASCII characters representing the double quotes around the term "AS IS".  In effect, this effectively terminated the string at the non-ASCII component and the input to Tcl_Eval became bad syntax.

The very easy solution is to replace the non-ASCII characters with escaped ASCII quote characters.

I'm not sure what it is about my system (Fedora, perhaps, or maybe just the version of Tcl/Tk) causes the issue to show up, but it's definitely one of those cases where the ounce of prevention is worth the pound of cure.

---Tim

jjcherry56, 2018-10-23:
nasty. not sure why it didn't barf on my end. fixed now.

closed 2018-10-23 by jjcherry56

jjcherry56, 2018-10-25:
I took another swipe at this with 'od -c' to see the non-printing characters. Let me know if it works this time.

RTimothyEdwards, 2018-10-25:
Looks okay now, thanks.