The-OpenROAD-Project/OpenSTA #3: openSTA crashes at link_design

kunalg123, 2018-10-05:
Hi
I am trying to run a very small design using openSTA, and looks like it crashes at link_design with segmentation fault message. Can you please have a look? 
Steps to reproduce the crash
1) Download testcase.tar.gz
2) tar -xvcf testcase.tar.gz
3) cd testcase
4) ./apps/sta -f map9v3.cmd

jjcherry56, 2018-10-23:
where is testcase.tar.gz hiding?

kunalg123, 2018-10-25:
Hi 
Thanks for the reply
This issue was raised 20days ago and in meantime we have fixed this issue. Would appreciate to have an early reply for issues, as we are planning to use this for real chip tapeout

jjcherry56, 2018-10-25:
who is kunalg123?

kunalg123, 2018-10-25:
I really apologize, as I forgot to introduce myself
My name is Kunal Ghosh and myself & Anagha Ghosh run an online company called VLSI System Design (VSD) Corp. Pvt. Ltd., primarily focused on making online content promoting open-soruce EDA tools. 
Here's my company link:
[https://www.vlsisystemdesign.com/about-us/](https://www.vlsisystemdesign.com/about-us/)

I, myself have STA and Physical design background, while Anagha has Electrical back-ground and looks into marketing, operations and strategies. Currently we are working on RISC-V based full chip tape-out using all open-source EDA tools

jjcherry56, 2018-10-25:
I am new to github and didn't see your "issue" until two days ago. I get notifications now so new issues  won't sit for 20 days in the future.

kunalg123, 2018-10-26:
Thank You jjcherry
Really appreciate

closed 2018-10-26 by kunalg123

jjcherry56, 2018-10-26:
If you have a testcase that causes a seg fault I would like to see it, even if you have found a work around.