The-OpenROAD-Project/OpenSTA #4: set_input_transition -clock not supported

kunalg123, 2018-10-26:
Hi 
In some of our constraints file, we have below constraint
1) set_input_delay -clock  clk -min -rise  102 [get_ports  mem_rdata_0_ ]

But instead, openSTA uses below
2)set_input_delay -min -rise  102 [get_ports  mem_rdata_0_ ]

While using 1) OpenSTA gives an error that -clock option not supported. Any reason?
We wanted to keep consistent SDC constraints across different tools for benchmarking and evaluation. Can you please let know your thoughts?

jjcherry56, 2018-10-26:
set_input_delay -clock is supported. If you have a testcase that show otherwise you should submit it.

jjcherry56, 2018-10-26:
Note that the title of the issue does not match the command in the body of the issue.
set_input_transition -clock is not supported. It is basically meaningless because
transition times are merged. Separate transition times are not used for each clock
in delay calculation.

kunalg123, 2018-11-11:
Hi James
Did you had a chance to look into the syntax?
Do you want me to close this issue?

jjcherry56, 2018-11-11:
I understand the syntax.
Using the -clock argument is meaningless with graph based delay calculation
so
it flagged as unsupported. If you omit it you will get the same result,

closed 2018-11-11 by kunalg123