The-OpenROAD-Project/OpenSTA #5: library keyword 'revision' interpretation error

kunalg123, 2018-10-29:
Hi James,
In our memory libraries, we have below lines:
/* Library Documentation */
  date     : ".September xxxx"
  revision : .x.x
  comment  : "Unit Area representation == xx.xxx sq.micron" ; 

While reading these libraries, openSTA gives out below error while reading the "revision" keyword
**line 42 syntax error, unexpected $undefined, expecting FLOAT or STRING or KEYWORD or '!'.**

If I comment all above lines from library, openSTA works fine. 
Do you think, this can be fixed in source code? We have many such libraries, and commenting in each library is a somewhat tedious task

jjcherry56, 2018-10-29:
It looks like garbage in the liberty file to me.
revision is a string attribute and your example is not enclosed in double quotes.
I would say openSTA is liberal in its parsing, which is fine but not a standard I feel required to support.

kunalg123, 2018-10-30:
Alright
Let me inform the library team
Will close this issue for now

closed 2018-10-30 by kunalg123