The-OpenROAD-Project/OpenSTA #8: set_wire_load_model for instances

kunalg123, 2018-11-16:
Hi Jerry
I am using below command to set_wire_load_model, but its setting only for top and not for hierarchical instances
current_design mkSoc_wrapper
set_wire_load_model -name ForQA -library tsl18fs120_scl_ss
Is there a way I can set_wire_load_model to top and all instances as well?

jjcherry56, 2018-11-16:
When I run an example it annotates all hierarchical instances.

I suggest you construct a testcase with specific net references that you see
unannotated. google drive may be a reasonable way to do so.

As far as I am concerned english is a poor substitute for a testcase and I
require a testcase for any issue before I am even willing to comment on it.

There is no one named "Jerry" here. My name is James Cherry.

kunalg123, 2018-11-19:
Hi James
Really apologize for mis-spelling name. Corrected in main query
Also, I used below set of commands to set_wire_load_model for all instances, and it works. 
How have I confirmed whether it works? by looking into opensta timing reports and comparing with other tool timing reports. 
Can you please confirm, if this is the right way to do this?

current_design mkSoc_wrapper
set_wire_load_model -name 4000
current_instance
current_instance U1
set_wire_load_model -name 4000
current_instance
current_instance U1/DP_OP_3949J2_124_2298
set_wire_load_model -name 4000
current_instance

The only issue I see is missing "report_wire_load" command, which will confirm that wire_load values has been properly applied. I think this would be a good command to add. 
Is there a way/script to get list of all instances in design? Currently, I am using a local script, but it has a dependency on other tool to report instances

jjcherry56, 2018-11-19:
The only command in your "script" that has any effect is
"set_wire_load_model".
Scoping based on the current instance is not supported. It is applied
globally to
the design. In your example the wire_load model is the same in each case so
it is equivalent anyway.

You can use the "report_net -connections -verbose net_name" command to see
thed
annotations.

Don't hold your breath on "report_wire_load". In 18 years and 14 commercial
customers
the Parallax STA has never had a request for that command. wire load models
are so 1995.

To get all instances simply use the SDC command "get_cells *".

kunalg123, 2018-11-20:
Thanks James
We needed wire_load only for a pre-layout post-synth correlation activity with other tools, so that we benchmark opensta and use it for tapeout 
I will use your suggestions to see the annotations

closed 2018-11-20 by kunalg123