The-OpenROAD-Project/OpenSTA #10: read_parasitics warning - net not found

kunalg123, 2018-11-21:
Hi James
After reading parasitics, I get below opensta warning
opensta warning:
Warning: /home/anaghavsd/Desktop/work/to_kunal/mkSoc_wrapper.post_route.spef.max, line 1757581 net U1/lv_result__h66626\[5\] not found.

But when debugged more in SPEF file I see below:
Inside SPEF File:
*2233 U1/lv_result__h66626\[5\]

**Line no. 1757581** 78 *143:55 *2233:13 0.1392269

*D_NET *2233 4.270612

*CONN
*I *2234:Z O *L 0.0 *D dl03d1 *C 664.895 5310.3
*I *2235:A1 I *L 4.929 *C 668.2 5299.12
*N *2233:3 *C 664.28 5312.0
*N *2233:4 *C 664.28 5312.0
*N *2233:5 *C 664.28 5310.84
*N *2233:6 *C 664.28 5310.84
*N *2233:7 *C 666.52 5310.84
*N *2233:8 *C 666.52 5310.84
*N *2233:9 *C 666.52 5300.28
*N *2233:10 *C 666.52 5300.28
*N *2233:11 *C 668.2 5300.28
*N *2233:12 *C 668.2 5300.28
*N *2233:13 *C 668.2 5299.12

*CAP
1 *2234:Z 0.06939172
2 *2235:A1 0.0
3 *2233:3 0.1211474
4 *2233:4 0.06939172
5 *2233:5 0.2014812
6 *2233:6 0.1211474
7 *2233:7 0.2014812
8 *2233:8 0.6075613
9 *2233:9 0.2244984
10 *2233:10 0.6075613
11 *2233:11 0.2244984
12 *2233:12 0.101769
13 *2233:13 0.101769
14 *2233:12 *143:55 0.1392269
15 *2233:13 *143:55 0.1392269
16 *2233:8 *650420:9 0.4570534
17 *2233:10 *650420:9 0.4570534
18 *2233:8 *650438:264 0.0314983
19 *2233:10 *650438:264 0.0314983
20 *2233:10 *650438:263 0.03416146
21 *2233:10 *650438:265 0.09252438
22 *2233:3 *650438:264 0.1183352

*RES
1 *2233:3 *2233:4 6.000001
2 *2233:5 *2233:6 6.000001
3 *2233:7 *2233:8 6.000001
4 *2233:9 *2233:10 6.000001
5 *2233:11 *2233:12 6.000001
6 *2233:13 *2235:A1 6.000001
7 *2233:4 *2234:Z 0.1885714
8 *2233:6 *2233:3 0.4400001
9 *2233:10 *2233:8 3.125714
10 *2233:13 *2233:12 0.4400001
11 *2233:5 *2233:7 0.7485715
12 *2233:9 *2233:11 0.5885715

Opensta timing reports
**After loading the verilog, when I use the below commands, I do see the net. So where is the net not found? Am I missing anything?**
% get_nets U1/lv_result__h66626\[5\]
_60ace52700000000_p_Net
[... 81 lines cut ...]
         1     0.0049                                    U1/n83458 (net)                    
                          0.4280     0.0000    18.5490 v U1/U119807/I (bufbd4)              
                          0.0775     0.2558    18.8048 v U1/U119807/Z (bufbd4)              
         1     0.0118                                    U1/n119222 (net)                   
                          0.0775     0.0002    18.8050 v U1/U119809/I (buffd2)              
                          0.0611     0.1311    18.9362 v U1/U119809/Z (buffd2)              
         1     0.0053                                    U1/n119224 (net)                   
                          0.0611     0.0001    18.9363 v U1/U119808/I (buffd3)              
                          0.0733     0.2024    19.1387 v U1/U119808/Z (buffd3)              
         1     0.0033                                    U1/n119223 (net)                   
                          0.0733     0.0001    19.1388 v U1/dma_rg_cpa_5_reg_5_/D (sdnrb1)  
                                               19.1388   data arrival time                  

                                    20.0000    20.0000   clock CLK (rise edge)              
                                     6.7002    26.7002   clock network delay (propagated)   
                                    -0.9000    25.8002   clock uncertainty                  
                                     0.0000    25.8002   clock reconvergence pessimism      
                                               25.8002 ^ U1/dma_rg_cpa_5_reg_5_/CP (sdnrb1) 
                                    -0.4036    25.3966   library setup time                 
                                               25.3966   data required time                 
---------------------------------------------------------------------------------------------
                                               25.3966   data required time                 
                                              -19.1388   data arrival time                  
---------------------------------------------------------------------------------------------
                                                6.2578   slack (MET)

jjcherry56, 2018-11-21:
Yes, you are missing something important. A test case.

kunalg123, 2018-11-30:
Really Apologize for delay. Actually design is taped out and porting it to opensource libraries (which can be shared) just took some time 
Attaching testcase. You can check some of below nets
% get_nets U1/qspi1\$slave_rdata\[63\]
_e0bebf0700000000_p_Net
% get_nets U1/qspi1/v__h15039[17]  
_f06ed30700000000_p_Net

[](https://1drv.ms/u/s!Ai4WW_jutenggas652Vq2moOnyRA-g)

Steps to run testcase
1) Download
2) tar -xvzf testcase.tar.gz
3) cd testcase
4) sta -f mkSoc_wrapper.post_route.osu.cmd
Let me know if you face issue

jjcherry56, 2018-11-30:
These sort of issues are always name mapping / namespace related.

The problem is the spef is not consistent with the verilog.
For example, the first error reported is:

Warning: ./mkSoc_wrapper.post_route.osu.spef, line 928661 instance
U1/qspi1/s_xactor_f_rd_addr/D_OUT_reg\[0\] not found.

The verilog for the last instance in this path is:

DFFPOSX1 \s_xactor_f_rd_addr/D_OUT_reg[0] (.Q ( s_xactor_f_rd_addr$D_OUT[0]
)
    , .CLK ( CLK_cts_3_6 ) , .D ( n6706 ) ) ;

The escaped name means that the hierarchy dividers in the name are quoted.
In spef this corresponds to:

U1/qspi1/s_xactor_f_rd_addr\/D_OUT_reg\[0\]

IE, the last hierarchy divider needs to be escaped in spef because it
is escaped in verilog, for the same reason that the bus delimiters are
escaped. It may be a problem further upstream in the DEF.

SDC removes all quoting obscuring the issue. OpenSTA readers for other
formats
do not extend this promiscuity. I do not consider this a bug. It is a bug
in the spef writer.

jjcherry56, 2018-11-30:
Here are some other issues I saw in the testcase:

Since there is only one lib file it is also used for min (hold)
paths and specifying -max makes no sense:

  read_liberty -max osu018_stdcells.lib

Ditto for spef:

  read_parasitics -max ./mkSoc_wrapper.post_route.osu.spef

The -wavform argument is superfluous because the default is a 50%
duty cycle clock:

  create_clock -name TCK -period 20  -waveform {0 10} [get_ports CLK_tck]

This is just asking for trouble because some of the inputs are clocks,
and it makes no sense at all to specify an input delay on a clock.

  set_input_delay  2 -clock CLK [all_inputs]

You probably want the CLOCK to be propagated, not a specific PORT:

 set_propagated_clock [get_ports CLK]
 set_propagated_clock [get_ports CLK_tck]
 set_propagated_clock [get_clocks CLK]
 set_propagated_clock [get_clocks TCK]

So the followiing is simpler:

 set_propagated_clock [all_clocks]

crpr is enabled by default, so this command is not necessary:

  set sta_crpr_enabled 1

It is rather unusual to see crpr mode same_transition.
The default same_pin is more common.

  set sta_crpr_mode same_transition

The current_design command has no effect on the report_checks command
so there is no point in using it here.

  current_design mkSoc_wrapper

Note that no timing paths are reported.

  report_checks
  No constrained paths.

This is because the clocks run into "dummy" pad cells that have no timing
groups:

report_edges -from CLK
CLK -> clk/PAD wire
  r -> r 0.00|0.00:0.00|0.00
  f -> f 0.00|0.00:0.00|0.00
% report_edges -from clk/PAD
report_edges -from clk/PAD

% report_edges -from CLK
report_edges -from CLK
CLK -> clk/PAD wire
  r -> r 0.00|0.00:0.00|0.00
  f -> f 0.00|0.00:0.00|0.00
% report_edges -from clk/PAD
report_edges -from clk/PAD
% report_instance clk
report_instance clk
Instance clk
 Cell: pc3d21
 Library: tsl18cio150_max
 Path cells: pc3d21
% report_lib_cell pc3d21
report_lib_cell pc3d21
Cell pc3d21
Library tsl18cio150_max
File dummy.lib
 PAD input
 CIN output

looking at dummy.lib:

cell(pc3d21) {
pin(PAD) {
direction : input;}
pin(CIN) {
direction : output;}
}

you can see why the clock did not get very far.

kunalg123, 2018-12-04:
> IE, the last hierarchy divider needs to be escaped in spef because it is escaped in verilog, for the same reason that the bus delimiters are escaped. It may be a problem further upstream in the DEF.
> I do not consider this a bug. It is a bug in the spef writer.

Not sure why openSTA behaves that way
I read same spef and netlist in 2 other timing tools, and looks like, its getting annotated. report_annotated_parasitics from one of the tool is below
Anyways, if that's the case with openSTA, let me find out a way to dump spef's in a way openSTA understands. I will keep you posted
If you know of a way, where your other customers have resolved spef write issue, can you please help let us know?

                    |         |         |         |   RC    |   Not   |
Net Type            |  Total  | Lumped  |  RC pi  | network |Annotated|
--------------------+---------+---------+---------+---------+---------+
Internal nets       |         |         |         |         |         |
  - Pin to pin nets |  467222 |       0 |       0 |  467222 |       0 |
  - Driverless nets |   11241 |       0 |       0 |       0 |   11241 |
  - Loadless nets   |      71 |       0 |       0 |       0 |      71 |
--------------------+---------+---------+---------+---------+---------+
Boundary/port nets  |         |         |         |         |         |
  - Pin to pin nets |     136 |       0 |       0 |     136 |       0 |
  - Driverless nets |       0 |       0 |       0 |       0 |       0 |
  - Loadless nets   |       2 |       0 |       0 |       0 |       2 |
--------------------+---------+---------+---------+---------+---------+
                    |  478672 |       0 |       0 |  467358 |   11314 |

kunalg123, 2018-12-04:
Hi James
I used a fresh testcase, and taken care of escaped characters (I still find it difficult to accept what you mentioned about escaped character, as 2 other tools annotate the parasitics as is)
Below is the annotation report for the new testcase from one of other tools

                    |         |         |         |   RC    |   Not   |
Net Type            |  Total  | Lumped  |  RC pi  | network |Annotated|
--------------------+---------+---------+---------+---------+---------+
Internal nets       |         |         |         |         |         |
  - Pin to pin nets |  624306 |       0 |       0 |  624306 |       0 |
  - Driverless nets |   28457 |       0 |       0 |       0 |   28457 |
  - Loadless nets   |      71 |       0 |       0 |       0 |      71 |
--------------------+---------+---------+---------+---------+---------+
Boundary/port nets  |         |         |         |         |         |
  - Pin to pin nets |     136 |       0 |       0 |     136 |       0 |
  - Driverless nets |       0 |       0 |       0 |       0 |       0 |
  - Loadless nets   |       2 |       0 |       0 |       0 |       2 |
--------------------+---------+---------+---------+---------+---------+
                    |  652972 |       0 |       0 |  624442 |   28530 |

And opensta gives below error
read_parasitics -max /home/anaghavsd/Desktop/work/tools/vsdflow/vsdflow/qflow_standalone_testing/mkSoc_wrapper/pnr/osu018_yosys_synth_icc/mkSoc_wrapper.postroute.osu.spef.max
Warning: /home/anaghavsd/Desktop/work/tools/vsdflow/vsdflow/qflow_standalone_testing/mkSoc_wrapper/pnr/osu018_yosys_synth_icc/mkSoc_wrapper.postroute.osu.spef.max, line 178 syntax error, unexpected $undefined, expecting QSTRING or IDENT or NAME.

Attaching testcase in below link. Steps to reproduce remains same
[https://1drv.ms/u/s!Ai4WW_jutenggatKLBAtD_3CaOu_tQ](https://1drv.ms/u/s!Ai4WW_jutenggatKLBAtD_3CaOu_tQ)

jjcherry56, 2018-12-05:
The spef format is defined by an IEEE spec, not the behavior of a tool that
I have no faith in.
Maybe you should read the spec.
OpenSTA has been reading legitimate spef files for 18 years.

The spef file here conflates verilog escaping with spef escaping, which are
two completely different animals. Did you "fix" this with a perl script or
something?
I cannot believe that any commercial tool would write a spef file that
looks like that.

If you want to pay Parallax Software to add a -allow_brain_damaged_escaping
flag
to the read_parasitics file, we can arrange that.

kunalg123, 2018-12-05:
The first testcase SPEF was written by a commercial industry grade tool (which you mentioned its a spef write issue). This was read successfully using another commercial industry grade eda tool  

The second testcase SPEF was written by a commercial industry grade tool and I modified it using shell script to remove escaping. This was also read successfully using another commercial industry grade eda tool 

If you want, I can share the run logs for other tools over team viewer. I don't see any other way to solve this problem other than getting to the root of it. I think we should do a quick team viewer so I can show you LIVE steps. 

**Let me know if that works. I can setup a date and time for the call.** 
Intention is not to blame any tools here, but to solve an issue which me (and several people using openSTA here) are facing

kunalg123, 2018-12-05:
Hi James
I created another very small testcase for us to debug even more deeper. I see the issue again here (regarding what you said about escaped character). But what's surprising is openSTA does recognize such nets using get_nets. Then why doesn't it annotates. Below is one such example
There is also a case where a net is genuinely not there [% get_nets _113_\[6\]] which openSTA reports correctly (you can see list of not-annotated nets at bottom of this message). 

I guess below information should be good enough for your debug. Let me know if you need anything else. Will be happy to help
Attaching testcase:

[testcase3.tar.gz](https://github.com/abk-openroad/OpenSTA/files/2649334/testcase3.tar.gz)

% get_nets startbuf\[1\]
_0065940100000000_p_Net
% report_checks -through [get_nets startbuf\[1\]]
Startpoint: _306_ (rising edge-triggered flip-flop clocked by clk)
Endpoint: _274_ (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

     Delay       Time   Description
---------------------------------------------------------------
      0.00       0.00   clock clk (rise edge)               
      0.00       0.00   clock network delay (ideal)         
      0.00       0.00 ^ _306_/CLK (DFFSR)                   
      0.23       0.23 ^ _306_/Q (DFFSR)                     
      0.10       0.33 v _125_/Y (NOR2X1)                    
      0.13       0.45 v _246_/Y (AND2X2)                    
      0.00       0.46 v _274_/D (DFFSR)                     
                 0.46   data arrival time                   

     20.00      20.00   clock clk (rise edge)               
      0.00      20.00   clock network delay (ideal)         
      0.00      20.00   clock reconvergence pessimism       
                20.00 ^ _274_/CLK (DFFSR)                   
     -0.09      19.91   library setup time                  
                19.91   data required time                  
---------------------------------------------------------------
                19.91   data required time                  
                -0.46   data arrival time                   
---------------------------------------------------------------
                19.45   slack (MET)                         

**Annotation report from other tool**
                    |         |         |         |   RC    |   Not   |
Net Type            |  Total  | Lumped  |  RC pi  | network |Annotated|
--------------------+---------+---------+---------+---------+---------+
Internal nets       |         |         |         |         |         |
  - Pin to pin nets |     142 |       0 |       0 |     142 |       0 |
  - Driverless nets |       0 |       0 |       0 |       0 |       0 |
  - Loadless nets   |       0 |       0 |       0 |       0 |       0 |
--------------------+---------+---------+---------+---------+---------+
Boundary/port nets  |         |         |         |         |         |
  - Pin to pin nets |      38 |       0 |       0 |      12 |      26 |
  - Driverless nets |       0 |       0 |       0 |       0 |       0 |
  - Loadless nets   |       0 |       0 |       0 |       0 |       0 |
--------------------+---------+---------+---------+---------+---------+
                    |     180 |       0 |       0 |     154 |      26 |

**Nets not annotated from other tool**
Error: Cannot find net '_114_[7]' in design 'map9v3' 
Error: Cannot find net '_114_[6]' in design 'map9v3' 
Error: Cannot find net '_114_[5]' in design 'map9v3'
Error: Cannot find net '_114_[4]' in design 'map9v3' 
Error: Cannot find net '_114_[3]' in design 'map9v3' 
Error: Cannot find net '_114_[2]' in design 'map9v3' 
Error: Cannot find net '_114_[1]' in design 'map9v3' 
Error: Cannot find net '_114_[0]' in design 'map9v3' 
Error: Cannot find net '_111_[7]' in design 'map9v3'
Error: Cannot find net '_111_[6]' in design 'map9v3' 
Error: Cannot find net '_111_[5]' in design 'map9v3'
Error: Cannot find net '_111_[4]' in design 'map9v3' 
Error: Cannot find net '_111_[3]' in design 'map9v3'
Error: Cannot find net '_111_[2]' in design 'map9v3' 
Error: Cannot find net '_111_[1]' in design 'map9v3' 
Error: Cannot find net '_111_[0]' in design 'map9v3' 
Error: Cannot find net '_112_' in design 'map9v3' 
Error: Cannot find net '_113_[8]' in design 'map9v3' 
Error: Cannot find net '_113_[7]' in design 'map9v3' 
Error: Cannot find net '_113_[6]' in design 'map9v3' 
Error: Cannot find net '_113_[5]' in design 'map9v3'
Error: Cannot find net '_113_[4]' in design 'map9v3' 
Error: Cannot find net '_113_[3]' in design 'map9v3' 
Error: Cannot find net '_113_[2]' in design 'map9v3' 
Error: Cannot find net '_113_[1]' in design 'map9v3' 
Error: Cannot find net '_113_[0]' in design 'map9v3'

kunalg123, 2019-01-06:
Checked this with PEX spef. OpenSTA **still not able to read post-layout spef's** with hierarchy separator. Please let us and everyone know if there a plan to fix this as none of post-layout designs can run on openSTA. 

If anyone has hack or work-around for spef generation without hierarchy separator, can you please share? The first and third testcase can be used for debug/analysis/workaround

Ranjandv, 2019-01-28:
Hi Kunal,

Could you add in the DEF and the initial netlist which you used to start the PNR??

R

kunalg123, 2019-01-28:
Hi @Ranjandv 
I will have to recreate the entire database once more. Are you looking forward to reproduce the error? 
I will anyways, recreate the testcase entire testcase and upload

Ranjandv, 2019-01-29:
Any simple testcase with def would be good. I wanted to check the same thing in def also.
I wanted to understand more on the issue, seems like the issue is while generating the spef which might be coming from DEF.

R

Ranjandv, 2019-02-04:
Hi Kunal,

Would it be possible to add the def by this week??

Ranjan

tspyrou, 2020-09-14:
Closing, re-open if a testcase is available.

closed 2020-09-14 by tspyrou