The-OpenROAD-Project/OpenSTA #11: Liberty memory support

mguthaus, 2018-12-05:
The read_liberty command reads a cell library, but it does not recognize memories. These require support for the memory cell type. You can see an example from OpenRAM at:

https://github.com/VLSIDA/OpenRAM/blob/master/compiler/tests/golden/sram_2_16_1_scn3me_subm_TT_5p0V_25C.lib

Let me know if you have any questions or if you have suggestions for adjusting memory lib files for use in OpenSTA.

jjcherry56, 2018-12-05:
I fixed a stupid bug parsing the units in the referenced file but don't see anything that would prevent it from functioning correctly. The ports and timing groups are same as any other cell with busses. Is there some specific problem you are seeing?

mguthaus, 2018-12-06:
My apologies! I was making a stupid mistake. It does seem to work now.

closed 2018-12-06 by mguthaus