The-OpenROAD-Project/OpenSTA #12: read_liberty warning

kat54347, 2018-12-13:
Hi James,
I got the warnings below after executing read_liberty command.

> Warning: axis type normalized_voltage not supported.
> Warning: missing variable_2 attribute.

This is the lookup table which caused the warnings

```
lu_table_template (waveform_template_name) {
 variable_1: input_net_transition;
 variable_2: normalized_voltage;
 index_1 ("float…, float");
 index_2 ("float…, float");
 }
```

This variable is supported by Liberty format but may be unused here. 
Should I just ignore the warning messages like these? Thanks!

jjcherry56, 2018-12-13:
like it says, it is not supported.
if you depend on functionality related to that attribute it won't work.

kat54347, 2018-12-13:
ok. Thanks!

closed 2018-12-13 by kat54347