The-OpenROAD-Project/OpenSTA #13: aclocal warning

kat54347, 2018-12-13:
Hi James, 
When I tried to install OpenSTA and executed `$ ./bootstrap `, I got the error messages below:
```
aclocal: couldn't open directory `m4': No such file or directory
autoreconf: aclocal failed with exit status: 1
```

After I create a directory named m4 by `$ mkdir -p m4`,  the issue is solved.

mgwoo, 2018-12-13:
In my case, I've solved this issue by typing 

    $ libtoolize

and tried

    $ ./bootstrap

jjcherry56, 2018-12-13:
what version of autoconf do you have installed?
the first line is just a warning and "normal".
it works fine in autoconf 2.69, the required version documented in INSTALL

> autoreconf --install
aclocal: warning: couldn't open directory 'm4': No such file or directory
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:379: installing './compile'
configure.ac:570: installing './config.guess'
configure.ac:570: installing './config.sub'
configure.ac:12: installing './install-sh'
configure.ac:12: installing './missing'
app/Makefile.am: installing './depcomp'
configure.ac: installing './ylwrap'

kat54347, 2018-12-13:
The autoconf on my machine was 2.63, but I still got the same messages after I installed version 2.69 of autoconf. 
`$ libtoolize` before `$ ./bootstrap` do solve the problem.

kat54347, 2018-12-14:
It was 1.11.1 and 2.2.6b.
After I update them to 1.15.1 and 2.4.6, I got new error messages like below 
```
configure.ac:384: error: possibly undefined macro: AC_DISABLE_SHARED
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:576: error: possibly undefined macro: AC_PROG_LIBTOOL
autoreconf: /path/to/autoconf failed with exit status: 1
```
`$ libtoolize` solves the problem though.

jjcherry56, 2018-12-15:
I posted a release on github that has a tarfile that lets you build without Autotools. You would be better off using that method. See the updated INSTALL.

closed 2018-12-15 by jjcherry56