The-OpenROAD-Project/OpenSTA #16: report_checks with -from -through -to options

gguo4, 2019-03-18:
Dear developer,

I am using the report_checks command with -though and -to options to show top k critical paths involving though pins and destination pin. I can get k paths by using -endpoints k, but slacks of these paths are always INF if I choose to use the register input (commonly considered as a type of endpoint) as -to pin. I experimented this on a classic s27 design. The command I used is

`report_checks -unconstrained -format end -endpoint_count 4 -through inst_6/A1 -to inst_16/D`

I obtained the following result.

inst_16/D (DFFR_X2)                      INF  202.22     INF (MET)
inst_16/D (DFFR_X2)                      INF  181.02     INF (MET)
inst_16/D (DFFR_X2)                      INF  177.97     INF (MET)
inst_16/D (DFFR_X2)                      INF  153.33     INF (MET)

If I choose to use primary output pin G17 as -to pin. The slack values are valid.

G17 (output)                           -0.20  177.68 -177.88 (VIOLATED)
G17 (output)                           -0.20  174.15 -174.35 (VIOLATED)
G17 (output)                           -0.20  153.04 -153.24 (VIOLATED)
G17 (output)                           -0.20  138.33 -138.53 (VIOLATED)

I am not sure if I missed any steps or I used the wrong pin as -to pin. I have attached the entire design below. "s27.tcl" is the file I used as command file to OpenSTA and "opensta_prev.log" is the logfile I had when I ran the input command file.

[s27.zip](https://github.com/abk-openroad/OpenSTA/files/2980157/s27.zip)

jjcherry56, 2019-03-19:
Dear User,

You have some problems with your libraries.
s27_Early.lib is missing setup_rising/falling timing groups
s27_Late.lib is missing hold_rising/falling timing groups

The structure of the libraries must match exactly.

jjcherry56, 2019-03-19:
Looking closer the command script is pretty confused.
The read_liberty commands redefine the same library cells without
distinguishing
the min and max libraries. Do you think it is going to use the file name to
understand
which is which? It should look like this:

read_liberty -min $early_lib
read_liberty -max $late_lib

Again, the structure of the libraries file must be identical.
The only difference should be the numbers.
Perhaps an error or warning should be printed, but I've never seen this
sort of confusion.

There is no point in calling "find_timing -full". That is implicit in all
of the reporting commands.

It makes no sense to call "log_begin" twice.

gguo4, 2019-03-19:
Thanks a lot for your reply! I am using 
read_liberty -min $early_lib
read_liberty -max $late_lib
to distinguish between min and max libraries. However, I still get INF on slacks. Should I merge both min and max libraries so that the merged file has both setup_rising/falling and hold_rising/falling groups.

jjcherry56, 2019-03-19:
I'll say it a third time.
The libraries much be structurally the same.
Every port. Every timing group.
I am adding error messages to check them.
Normally libraries are generated by scripts that know better.

gguo4, 2019-03-19:
I can get slack values now. Thanks!

closed 2019-03-19 by gguo4

jjcherry56, 2019-03-21:
I commited checks that report warnings for libraries without matching timing groups.
Unfortunately, tau contest data is based on IBM's broken libraries that suffer from this problem.
I also added some example command files and libraries that will be used in future tau
contests that use the same cell names and are correctly formed.