The-OpenROAD-Project/OpenSTA #17: License potential issue

just22, 2019-03-23:
Hello,

there is an on-going effort to import OpenSTA in the OpenBSD port tree. Currently the main concern is related to the software license; even if it seems clear from all file headers that you are using the GPLv3, the "License" paragraph in README.md is confusing:

> Copyright (c) 2019, Parallax Software, Inc. All rights reserved.
> 
> No part of this document may be copied, transmitted or disclosed in any form or fashion without the express written consent of Parallax Software, Inc.

Could you please clarify? Does your software integrally adhere to the GPL?

Thanks and regards

--
Alessandro

jjcherry56, 2019-03-23:
Well that would only apply to the README.md file, since that's the file
that has the notice, now wouldn't it?
I have filters generate the open source version from the proprietary
version.
Obviously not all the bug are worked out of them.

just22, 2019-03-26:
Hello James,

> Well that would only apply to the README.md file, since that's the file
> that has the notice, now wouldn't it?

Well, README.md makes part of the distribution tarball and that notice would prevent to put the distfile on public FTP mirrors. Anyway, I see that with commit 20ab0a1f235af6f1896c25193228adefd6e7fe2c you amended the copyright for both README.md and seach/Genclks.hh (the only other file that had a similar problem), thanks for that.

I think you can close this ticket.

Regards

--
Alessandro

closed 2019-03-29 by just22