The-OpenROAD-Project/OpenSTA #18: Request change in CMakeLists.txt so it works with Mac Ports installation of CUDD

geochrist, 2019-03-24:
Hi James,

I'm trying to compile OpenSTA on MacOS with Mac Ports installation of libraries -- in particular CUDD.  I found my MacPort installation of cudd located at /opt/local/include/cudd (for headers) and /opt/local/lib/cudd (for the library).

I had to modify CMakeLists.txt to be:
Because I am using MacPorts, the CUDD package is installed in /opt/local, with include files in /opt/local/include/cudd and library at /opt/local/lib/cudd/libcudd*.

To allow me to set CUDD_INCLUDE, I needed to modify CMakeLists.txt to read:

if("${CUDD}" STREQUAL "" OR "${CUDD}" STREQUAL "0")
  set(CUDD_INCLUDE "")
  set(CUDD_LIB "")
  set(CUDD_FOUND FALSE)
  set(CUDD 0)
  message(STATUS "CUDD library: not found")
else()
  if("${CUDD_INCLUDE}" STREQUAL "")   <<==changed lines
    set(CUDD_INCLUDE ${CUDD}/include)  <<==changed lines
  endif()                              <<==changed lines
  find_library(CUDD_LIB NAMES cudd PATHS ${CUDD}/lib)
  if (CUDD_LIB)
     # CUDD referenced by StaConfig.hh.cmake
    set(CUDD 1)
    message(STATUS "CUDD library: ${CUDD_LIB}")
    message(STATUS "CUDD header: ${CUDD_INCLUDE}")
  endif()
endif()

Then, with the command line for cmake: 

cmake .. -DCUDD_LIB=/opt/local/lib/cudd/libcudd.dylib -DTCL_LIB=/opt/local/lib/libtcl.dylib -DZLIB_ROOT=/opt/local/lib/libz.dylib -DCUDD_INCLUDE=/opt/local/include/cudd

I was able to pass the "missing cudd header file" and build "sta".

George

jjcherry56, 2019-03-25:
I changed the cmake script to look in $CUDD/lib/cudd etc so it should work
with your ports install.

geochrist, 2019-03-25:
Hi James,

I think something else broke.  I’m now getting:

aragorn:OpenSTA/OpenSTA/build] gjchen% make
/opt/local/bin/cmake -S/Users/gjchen/src/OpenSTA/OpenSTA -B/Users/gjchen/src/OpenSTA/OpenSTA/build --check-build-system CMakeFiles/Makefile.cmake 0
/opt/local/bin/cmake -E cmake_progress_start /Users/gjchen/src/OpenSTA/OpenSTA/build/CMakeFiles /Users/gjchen/src/OpenSTA/OpenSTA/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/OpenSTA.dir/build.make CMakeFiles/OpenSTA.dir/depend
[  1%] [FLEX][VerilogLex] Building scanner with flex 2.6.4
cd /Users/gjchen/src/OpenSTA/OpenSTA && /opt/local/bin/flex --prefix=VerilogLex_ --header-file=/Users/gjchen/src/OpenSTA/OpenSTA/verilog/VerilogLex.hh -o/Users/gjchen/src/OpenSTA/OpenSTA/verilog/VerilogLex.cc verilog/VerilogLex.ll
[  1%] Generating ../app/StaApp_wrap.cc
cd /Users/gjchen/src/OpenSTA/OpenSTA && /opt/local/bin/swig -tcl8 -c++ -namespace -prefix sta -o /Users/gjchen/src/OpenSTA/OpenSTA/app/StaApp_wrap.cc /Users/gjchen/src/OpenSTA/OpenSTA/app/StaApp.i
cd /Users/gjchen/src/OpenSTA/OpenSTA && /Users/gjchen/src/OpenSTA/OpenSTA/etc/SwigCleanup.tcl /Users/gjchen/src/OpenSTA/OpenSTA/app/StaApp_wrap.cc
make[2]: *** No rule to make target `../tcl/Compatibility.tcl', needed by `../app/TclInitVar.cc'.  Stop.
make[1]: *** [CMakeFiles/OpenSTA.dir/all] Error 2
make: *** [all] Error 2

I’m pretty sure I flushed my changes and pulled the latest code down.

I noticed the CMakeLists.txt differ from previous version by the addition of tcl/Compatibility.tcl to the STA_TCL_FILES set.  Did you check that file into git?

Thanks.

George

geochrist, 2019-03-25:
P.S.  Here’s my pull result:

[aragorn:~/src/OpenSTA/OpenSTA] gjchen% git pull
Updating 20ab0a1..37df007
Fast-forward
 CMakeLists.txt    | 645 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-------------------------------------------------------------------
 README.md         |  29 +++++--
 search/Genclks.hh |  16 ----
 3 files changed, 352 insertions(+), 338 deletions(-)

jjcherry56, 2019-03-25:
> AhI8lfFdFSVQ4KxF6ag5bTas2C6Opx5Aks5vZzzRgaJpZM4cFYu2 <https://github.com/

try it now

geochrist, 2019-03-25:
Working now!  Thanks.

George

closed 2019-03-25 by geochrist