The-OpenROAD-Project/OpenSTA #19: multiple report_checks command

gguo4, 2019-03-25:
Dear developer,

I am trying to report multiple timing paths in different restrictions
`report_checks -through PIN1 -through PIN2 -through PIN3`
`report_checks -through PIN4 -through PIN5 -through PIN6`
...

I am stacking multiple `report_checks` in my command file, but I've got "Search.cc:2669 max tag group index exceeded" and "Segmentation fault" when the number of `report_checks` is over 7000. I would like to know the proper way to write my command file without getting this error.

Thanks

jjcherry56, 2019-03-25:
I can't respond without a testcase.

gguo4, 2019-03-25:
My bad. This is the command file I used. I get obtain correct results for the previous 7422 queries, but it crashes with the error I mentioned afterwards.

[vga_lcd.txt](https://github.com/abk-openroad/OpenSTA/files/3001785/vga_lcd.txt)

jjcherry56, 2019-03-25:
A command file is not a test case.
A test case is all of the files required to reproduce the bug.
It so happens that I have vga_lcd because it is an old tau test case
and that seems to be the only data floating around the universities
so I was able to reproduce this.

If your goal is to get the results you can insert a "find_timing -full"
every 1000 lines
to "garbage collect". I did this with a emacs macro.
If your goal is to break OpenSTA, congratulations. I have some recycling to
attend to.

gguo4, 2019-03-26:
Thanks! This is exactly what I needed. It runs smoothly right now. 
I have another question to ask. I am wondering if changing the number of threads when I ran OpenSTA will improve the performance for this sort of workload.

jjcherry56, 2019-03-26:
threads will help, but won't fix the underlying problem, which is cleaning
up the
garbage left by report -from/-thru/-to. I'll get to it but not today.

gguo4, 2019-03-26:
It's good for me as long as I can obtain the results. Thanks again for your answers.

closed 2019-03-26 by gguo4

jjcherry56, 2019-03-27:
I pushed a fix so you don't need to modify your scripts.

jjcherry56, 2019-03-29:
I am curious what the point of 46,000 timing reports is.

gguo4, 2019-03-29:
It's from the TAU 2018 contest. I think they did it for the contest purpose.