The-OpenROAD-Project/OpenSTA #21: Missing util/StaConfig.hh in install file list

just22, 2019-03-28:
Hello,

trying to build a port which uses the OpenSTA API (TritonSizer), I noticed that the header file `util/StaConfig.hh` is missing in the list of files to be installed (this breaks the code in multiple places).

I solved with the following simple patch:

```
Index: CMakeLists.txt
--- CMakeLists.txt.orig
+++ CMakeLists.txt
@@ -366,6 +366,7 @@ set(STA_HEADERS
   util/ReportStd.hh
   util/ReportTcl.hh
   util/Set.hh
+  util/StaConfig.hh
   util/Stats.hh
   util/StringSeq.hh
   util/StringSet.hh
```

jjcherry56, 2019-03-28:
looks good to me. I pushed the change to github.

just22, 2019-03-29:
Hello James,

thanks. But now the copyright notice in `CMakeLists.txt` is the wrong one...

--
Alessandro

jjcherry56, 2019-03-30:
ok, try again

just22, 2019-03-30:
Thanks James, problem solved. Closing.

closed 2019-03-30 by just22