The-OpenROAD-Project/OpenSTA #26: Crash when issuing report_power

byzantic, 2019-06-21:
OpenSTA crashes (segfault) when issuing report_power

The behaviour varies with the commit version. It occurs with commit e2dd765b9b4ba091db65291a54f6ed92f4a2603f, (as at 21/06/2019)

**does not** occur with a version I downloaded from master.zip on 18/06/19

**does** occur with a version from around (25/05/19)

**does not** occur with a version from around (29/01/19)

.. all the various versions are due to the fact that I am using docker images, and when building a docker image for other platforms, I pull the master.zip file from github (I should probably rethink my strategy!)

I'm having a problem pinning the versions down to a specific commit - is the commit number in a string anywhere in the code?

jjcherry56, 2019-06-21:
The specific version doesn't matter; I'm sure the head will crash based on
what I see here.
More important is either a back trace (which is probably not something you
can do if you
are using docker images because you need to recompile with
CMAKE_BUILD_TYPE=DEBUG),
or a testcase which reproduces the crash. I can kind of guess what is going
on but I think it depends
on specific pins missing on instances with specific internal power groups
in the liberty.

I doubt if you can see git hashes if you use zip files to get the sources.
Is there some reason you don't just clone the repo and use pulls to build
the docker images so you know where the came from?

byzantic, 2019-06-22:
Well, I nuke the source tree after building to reduce image size, so actually there's no record of the original code anyway. If I clone from git, but then delete it, that has the same effect..

But I'll have to rethink that strategy. I could clone from git, then squirrel away the hash using something like `git rev-parse --short HEAD > versionfile.txt`

jjcherry56, 2019-06-24:
If it doesn't crash with the latest version I consider it fixed.

closed 2019-06-24 by jjcherry56

byzantic, 2019-06-24:
.. sounds great .. err can't test it until tomorrow (GMT), but I'll let
you know asap!