The-OpenROAD-Project/OpenSTA #28: STA parser does not support nets name with "slashs"

egiacomin, 2019-07-19:
Hello,
Currently using the OpenROAD flow but I get a warning during the STA Placement step:

-----------------------
OpenSTA Copyright (c) 2018, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
INFO:  Now linking: grid_clb
INFO:  Successfully linked: grid_clb
Warning: , line 1 net grid_clb_0_\\/fle_2__ff_phy_1__D_0_ not found.
-----------------------

My Verilog was a structural Verilog design which I fed into Synopsys DC in order to get a flat design.
As such, many net names have "/" in their name. I replaced those by underscore and it seems to alleviate the issue. Is the parser not capable of dealing with "/" characters?

mgwoo, 2019-07-19:
Hi egiacomin, 

This issue is for TD-RePlAce, not for OpenSTA.
The warning you're facing is not a problem at all. 
(I still have to learn about SPEF delimiter's rule in OpenSTA. I wrote a simple checker whenever OpenSTA is instantiated in [here](https://github.com/The-OpenROAD-Project/RePlAce/blob/5cda3b0a6a99d95a97e645ac6130d48ece55b9fd/src/timingSta.cpp#L477-L485) (inside RePlAce))

I'll remove such a warning in the near future.  
(RePlAce would be rewritten in the near feature)

egiacomin, 2019-07-19:
Thanks for the quick reply. I'll ignore those warnings for now then.

closed 2019-07-19 by egiacomin