The-OpenROAD-Project/OpenSTA #31: OpenSTA cannot parse assigns to concatenated signals

byzantic, 2019-08-12:
I have just come across an issue with Yosys, where it produces assigns like the following:

`assign {a,b} = {c,d};`

OpenSTA cannot parse these. 

Now, the simple approach would be to stop Yosys generating such assigns, since they could always be separated into multiple assign statements. I think the issue has been raised before with Clifford, and he points out that it is legal verilog. I will raise it with him again, though.

Fixing the OpenSTA parser  would be great, but I suspect that might be a tricky job ..

Unfortunately, it is quite difficult to generate a simple example; whether or not such constructs are generated is highly dependent on the steps used in yosys.

jjcherry56, 2019-08-16:
It shouldn't be that hard to fix the OpenSTA parser. I won't be able to get to it for a while though.
It would help to have a specific example that shows where a/b/c/d are (inputs/outputs/gate input/gate/output). All you have to do is catch a verilog file produced by yosys and save it once.

byzantic, 2019-08-19:
Thanks, I'll put one up when I next get the problem.

vperrin59, 2020-05-16:
I fell into the exact same issue.
I can upload the verilog netlist here 
`assign { _09552_[2], _09552_[0] } = { _09475_[2], _09475_[0] };`

And the error in OpenSTA: 
Error: ../yosys/synth.v, line 72936 syntax error, unexpected '{', expecting ID.

macd, 2020-11-09:
I ran into this as well.  Looking at the Yosys Verilog writer, it looks like it defaults to using this style of Verilog.  As pointed out, this is legal Verilog and it is also produced by the commercial tools as well.  However, while I haven't extensively tested this, there may be a workaround.  For my design, I just used the "splitnets -ports" command before writing out the Verilog, and these kinds of assigns were not present in the netlist and the design loaded correctly into OpenSTA.

macd, 2020-12-03:
As of last week (I think) Yosys now has a flag for the Verilog writer to avoid concats on the lhs.  You can use this instead of doing a `splitnets` on the design.  Use  `write_veilog -simple-lhs xxx.v`

jjcherry56, 2021-01-19:
fixed in c13383fb issue#31 verilog concat assign

closed 2021-01-19 by jjcherry56