The-OpenROAD-Project/OpenSTA #32: [TritonSizer/OpenSTA] ./bootstrap not found

speedbooster, 2019-09-14:
Hi,
I had originally posted the issue under TritonSizer ([here](https://github.com/The-OpenROAD-Project/TritonSizer/issues/6)), but i just realized its likely an OpenSTA issue.

On CentOS7, upon **make**, it gives:

> Making directory /home/user/Downloads/openflow/TritonSizer/module/tcl/unix/install-sp/man/man3
> Making directory /home/user/Downloads/openflow/TritonSizer/module/tcl/unix/install-sp/man/mann
> Installing and cross-linking top-level (.1) docs
> Installing and cross-linking C API (.3) docs
> Installing and cross-linking command (.n) docs
> make[1]: Leaving directory '/home/user/Downloads/openflow/TritonSizer/module/tcl/unix'
> cd module/OpenSTA && mkdir -p install-sp &&
> ./bootstrap && ./configure --prefix=/home/user/Downloads/openflow/TritonSizer/module/OpenSTA/install-sp
> --with-lib=/home/user/Downloads/openflow/TritonSizer/module/tcl/unix/install-sp/lib
> --with-include=/home/user/Downloads/openflow/TritonSizer/module/tcl/unix/install-sp/include
> --with-tcl=/home/user/Downloads/openflow/TritonSizer/module/tcl/unix/install-sp/lib/tcl8.4 &&
> make install;
> /bin/sh: line 1: ./bootstrap: No such file or directory
> make: *** [Makefile:14: sta] Error 127

There s no bootsrap file in OpenSTA directory

jjcherry56, 2019-09-14:
OpenSTA no longer builds using autotools. TritonSizer either has to build using a version before 2/16/2019 or update the build to follow the instructions in README.md.

speedbooster, 2019-09-15:
i am only using cmake and make

closed 2019-09-15 by jjcherry56

speedbooster, 2019-09-15:
You closed this without a solution?