The-OpenROAD-Project/OpenSTA #33: Not able to read spef in openSTA

rajatbansal36, 2019-09-16:
Hi,
While i am giving a .spef file to openSTA, it shows error saying pin name not found.?
"Warning: ./synthesis/FIFO.mod.spef, line 1499 net CLK_bFbuf5 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1545 net CLK_bFbuf4 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1591 net CLK_bFbuf3 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1641 net CLK_bFbuf2 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1689 net CLK_bFbuf1 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1737 net CLK_bFbuf0 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1802 pin 61_1 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1803 pin 61_2 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1804 pin 61_3 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1805 pin 61_4 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1806 pin 61_5 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1807 pin 61_6 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1808 pin 61_7 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1809 pin 61_8 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1810 pin 61_9 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1811 pin 61_10 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1812 pin 61_11 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1813 pin 61_12 not found.
Warning: ./synthesis/FIFO.mod.spef, line 1814 pin 61_13 not found."

Above is the some portion of error at last the variable flag shows "1", & upon report_checks on same path, the arrival time decreases, which is expected to be increased.

Please let me know if i am processing something wrong in the procedure, else help me out of it.

jjcherry56, 2019-09-16:
What is your affiliation?
With no test case it is impossible to comment.

jjcherry56, 2022-07-13:
No testcase, no help.

closed 2022-12-26 by jjcherry56