The-OpenROAD-Project/OpenSTA #38: TCL_HEADER not included in CMakeList.txt includes

andyfox-rushc, 2020-01-15:
cmake option -DTCL_HEADER would appear not  to be working because TCL_HEADER not included in STA_INCLUDE_DIRS in CMakeLists.txt. Possible fix in CMakeLists.txt is:

set(STA_INCLUDE_DIRS
  app
  dcalc
  graph
  liberty
  network
  parasitics
  sdc
  sdf
  search
  util
  verilog
  ${CUDD_INCLUDE}
#Add Tcl header..  
  ${TCL_HEADER}  
  )

jjcherry56, 2020-01-16:
The README.md needs to be updated to reflect a recent change to the cmake
file.
You want to set TCL_INCLUDE_PATH instead (look further down the file).

jjcherry56, 2020-01-16:
hold on, that's not right.
definitely should not be adding it to STA_INCLUDE_DIRS but it isn't respecting it as an option
(it usually just works so I don't pay attention to the broken case)

jjcherry56, 2020-01-16:
Ok, I don't see a problem after trying it.
Can you be more specific about the problem and less specific about your proposed fix?

closed 2020-01-21 by jjcherry56