The-OpenROAD-Project/OpenSTA #42: Please consider moving headers into a subdirectory

yurivict, 2020-03-23:
The project installs 164 files into /usr/local/include. It's better if they would go into a dedicated subdirectory, for example /usr/local/include/opensta. Fewer files would be in include/, and the possibility of conflicts with other packages would also be lower.

jjcherry56, 2020-03-23:
good idea. nobody really uses make install, but that makes sense.
I'm convinced the library or headers even belong in the install target.
950b7c3 master origin/master mv install headers in /opensta, missing headers

closed 2020-03-23 by jjcherry56

yurivict, 2020-03-23:
> nobody really uses make install

The Arch Linux packages uses ```make install```: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=opensta-git

The FreeBSD package also uses ```make install```.

I'm sure all other packages too: https://repology.org/project/opensta/versions

jjcherry56, 2020-03-23:
I mean actual code users, not generic package installers.
20 years of users pre-github and I doubt they ever used make install.

yurivict, 2020-03-23:
Yes, users are supposed to install software through packages these days, and packages use ```make install```.