The-OpenROAD-Project/OpenSTA #50: Unable to source or read files

charumaurya, 2020-07-22:
Whenever I'm sourcing the examples files, or reading them, with the command
OpenSTA> source example1.tcl

I'm getting error : cannot read this file
Same while reading

rovinski, 2020-07-22:
Is the file in the same directory and does it have read permissions? If yes, please provide more details.

- OS name and version
- tcl version
- OpenSTA commit
- Steps to reproduce

I can't foresee this being an OpenSTA issue. `source` is a tcl built-in.

charumaurya, 2020-07-22:
@rovinski 
Hey, thanks but I figured it out myself!

Also, I had one doubt regarding the documentation provided as OpenSTA.pdf
I'm unable to execute the commands in the command line interpretor.
For eg. My script exam1.tcl contains: 
read_liberty example1_slow.lib
read_verilog example1.v
link_design top
read_sdf...
create_clock -name clk -period 10 {clk1 cl2 clk3}
set_input_delay -clock clk 0 {in1 in2}
all_registers -clock clk1

Should return the register connected to clk1.
But it gives no output.

rovinski, 2020-07-22:
Sourced scripts in tcl do not echo output by default. Try this instead:
```
puts "[all_registers -clock clk1]"
```

@charumaurya these are tcl issues not OpenSTA issues. Please only file an issue if you can verify there is a bug in OpenSTA itself. If you have help questions please use the [OpenROAD Gitter](https://gitter.im/The-OpenROAD-Project/community) instead.

closed 2020-07-22 by rovinski