The-OpenROAD-Project/OpenSTA #51: Missing proc get_pins_error

20Mhz, 2020-08-04:
Reference to missing proc at:
`tcl/Power.tcl:250    set ports [get_pins_error "pins" $keys(-pins)]`

Trying to use set_power_activity, but getting following error:
```
get_pins_error
invalid command name "get_pins_error"
```

jjcherry56, 2020-08-05:
fixed in commit
ae3da179 set_power_activity -pins

closed 2020-08-05 by jjcherry56

20Mhz, 2020-08-05:
Thank you, will test

20Mhz, 2020-08-06:
It seems the UI is in place now, but its not clear to me if the activity is getting through, as I see no change in reported power. 
I created the following example as test:

```
set flop [get_name [lindex [get_cells -of_objects [get_nets clk]] 0] ]
set clock_port [get_name [get_ports -of_object [get_net clk] -filter "direction == input"]]
puts "Testing Power on flop ${flop} for clock ${clock_port}"
foreach i {0 1 2 3 4 5 6 7 8 9 10} {
	# Set activity and report Power
	set_power_activity -input_port ${clock_port} -activity [expr 0.1 *[expr $i*$i]] -duty 0.5
	with_output_to_variable r "report_power -instance ${flop} -digits 7"
	# grep cell name
	foreach l [split $r '\n'] { if { [regexp ${flop} $l ] } { puts $l} }
}
```
No change:
```
% source test_power.tcl
Testing Power on flop _22045_ for clock clk
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
 7.4284026e-07 8.3394561e-08 1.9302500e-13 8.2623501e-07 _22045_
```

jjcherry56, 2020-08-06:
Without a complete test case I can't really say much, but it appears you
are trying to set the activity on clock and clocks activity is derived from
create_clock so I'm not surprised the result doesn't change.

jjcherry56, 2020-08-06:
And this example does not use the -pins argument to set_power_activity so
it is completely unrelated.

20Mhz, 2020-08-06:
Thanks for the insight, I've confirmed this works by constraining the non clock  cell pins instead [test_power.tcl](https://github.com/20Mhz/issues/blob/master/test_power/test_power.tcl)

```
Testing Power on flop _22045_ for clock clk
 7.2544111e-07 0.0000000e+00 1.9302500e-13 7.2544128e-07 _22045_
 7.3199857e-07 0.0000000e+00 1.9302500e-13 7.3199874e-07 _22045_
 7.5167088e-07 0.0000000e+00 1.9302500e-13 7.5167105e-07 _22045_
 7.8445817e-07 0.0000000e+00 1.9302500e-13 7.8445834e-07 _22045_
 8.3036025e-07 0.0000000e+00 1.9302500e-13 8.3036042e-07 _22045_
 8.8937730e-07 0.0000000e+00 1.9302500e-13 8.8937747e-07 _22045_
 9.6150927e-07 0.0000000e+00 1.9302500e-13 9.6150950e-07 _22045_
 1.0467561e-06 0.0000000e+00 1.9302500e-13 1.0467563e-06 _22045_
 1.1451178e-06 0.0000000e+00 1.9302500e-13 1.1451181e-06 _22045_
 1.2565945e-06 0.0000000e+00 1.9302500e-13 1.2565947e-06 _22045_
 1.3811859e-06 0.0000000e+00 1.9302500e-13 1.3811862e-06 _22045_
```