The-OpenROAD-Project/OpenSTA #52: unset_output_delay asks for unknown argument

quentinw91, 2020-08-17:
Hi,
I wrote the following constraint.sdc for gcd example (nangate45):
```
create_clock [get_ports clk]  -name core_clock  -period 2

#set output delay for all
set_output_delay -clock core_clock -max 1 [all_outputs]
set_output_delay -clock core_clock -min 0 [all_outputs]

#except for one...
unset_output_delay [get_ports resp_val]
set_output_delay -clock core_clock -max 0.5 [get_ports resp_val]
set_output_delay -clock core_clock -min 0   [get_ports resp_val]
```
When running opensta, I receive the following error :
...
Error: unset_output_delay requires two positional arguments.
number instances in verilog is 272
...

I also tried (even though it is marked as optional in the manual)
```
unset_output_delay -clock core_clock [get_ports resp_val]
```
but same response.
What am i missing ?
Thanks,
Quentin

jjcherry56, 2020-08-17:
You aren't missing anything; it is a bug. Fixed in 
17eced5f unset_input_delay/unset_output_delay no delay arg

closed 2020-08-17 by jjcherry56

jjcherry56, 2020-08-17:
note that you can also use [delete_from_list [all_ports] [get_ports resp_val]]

quentinw91, 2020-08-18:
ok thank you !