The-OpenROAD-Project/OpenSTA #54: make failed using tcl8.6 on ubuntu platform

simintao, 2020-09-09:
[  0%] Generating StaTclInitVar.cc
cd /cygdrive/e/Opensource/OpenSTA && etc/TclEncode.tcl /cygdrive/e/Opensource/OpenSTA/build_linux/StaTclInitVar.cc tcl_inits tcl/Util.tcl tcl/Graph.tcl tcl/Liberty.tcl tcl/Link.tcl tcl/Network.tcl tcl/NetworkEdit.tcl tcl/Sdc.tcl tcl/Search.tcl tcl/Cmds.tcl tcl/Variables.tcl tcl/Sta.tcl tcl/Power.tcl tcl/Splash.tcl dcalc/DelayCalc.tcl parasitics/Parasitics.tcl sdf/Sdf.tcl verilog/Verilog.tcl
": no such file or directoryog.tcl
    while executing
"open $filename r"
    (procedure "encode_file" line 2)
    invoked from within
"encode_file $filename"
    ("foreach" body line 2)
    invoked from within
"foreach filename $init_filenames {
  encode_file $filename
}"
    (file "etc/TclEncode.tcl" line 75)

jjcherry56, 2020-09-17:
build_linux is not part of OpenSTA.
Please provide a complete log of your build attempt.
It looks like you are trying to build from the wrong working directory.

simintao, 2020-12-24:
fix, It is windows tcl file can not be read on linux

closed 2020-12-24 by simintao