The-OpenROAD-Project/OpenSTA #58: Units getting lost in rewriting of SDC file.

arlpetergadfort, 2020-10-26:
When you specify the units in the SDC file (say in ns for time) and the library is in (ps), the output of the write_sdc command will write a clock as 5ps instead of 5ns (because it doesn't output the units or scale the information correctly).

One possible solution is to simply write the units used to the new SDC file, so ensure there is no loss of information.
I've created a patch that writes a section the output SDC file to address this issue:
[OpenSTA-writeunits.zip](https://github.com/The-OpenROAD-Project/OpenSTA/files/5439079/OpenSTA-writeunits.zip)

Thanks,
Peter

jjcherry56, 2020-10-26:
SDC units are specified in the liberty file. There is no SDC command to set them. set_cmd_units is an opensta specific command that is not part of SDC so your script writes invalid SDC.

closed 2020-10-26 by jjcherry56

arlpetergadfort, 2020-10-27:
I know you closed this, but I think there is a case to be made that if we use the set_cmd_units commands in opensta and then write the SDC file, it should correct the units in the output file from 5ns to 5000ps. It would not break the SDC format to correct the units (ie. scale the data correctly when writing).