The-OpenROAD-Project/OpenSTA #61: Re-appearance of NaNs in power

arlpetergadfort, 2020-11-03:
Sorry to recreate the issue. When compiling with 983149a4e42da16689ca1f82d886e2eb427948f3, everything works.
However, after bdd74687745ec6b031c6b05af0368ee006a85662, the `NaN`'s are back.
Going through the code, it would appear that issue is here:
https://github.com/The-OpenROAD-Project/OpenSTA/blob/c43b426d49246f9f0646f89bdc406a65b649b83d/search/Power.cc#L816
duty_sum_iter->second returns 0.0, which is causing the weight to become `NaN` again.
Wrapping the duty sum in a similar fashion corrects the issue: `if (isnormal(duty_sum)) weight = duty / duty_sum;`

I tested the fix [nan2.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/5483493/nan2.txt) on c43b426 and it correctly suppresses the `NaN`'s.

Thanks for your help,
Peter

jjcherry56, 2020-11-03:
Sorry about that. I got a little carried away with rewriting.. should work now

arlpetergadfort, 2020-11-03:
Thanks. I just pulled / recompiled / re-ran my design through and it is working again.

Thanks for all your help, 
Peter

closed 2020-11-03 by jjcherry56