The-OpenROAD-Project/OpenSTA #63: clock gate does not correctly impact report_power

arlpetergadfort, 2020-11-16:
I was looking at using the `set_case_analysis` to get some power numbers for different configurations of our design, but I noticed that settings settings that should have turned off a clock gate was not impacting the power analysis.
I also tried to use the `set_power_activity` but that didn't seem to make a difference, so the example I gave use `set_case_analysis`.

When looking through the power debugging, I noticed the registers that should have been gates had a lot of switching activity power.
The clock for those registers still had activity propagated to them, despite the gate being disabled.
Secondly, looking at the clock gate itself, the output pin also had activity on it, even when the `enable` pin was set to always disabled.

I've attached an example that contains two clock gates and two registers [clock-gating.zip](https://github.com/The-OpenROAD-Project/OpenSTA/files/5549782/clock-gating.zip).
In theory, if the first clock gate is disabled, there should be no activity in the module except for the input pins and if the second gate is disabled, only one of the two registers `r2` will be gated.

When looking through the example, you can see that setting the `set_case_analysis 0 g2/E` on the second clock gate (which should disable the clock on the second register), lowering the power for `r2` actually increases the power for `r2`.

- `r2` power without changing anything: 3.95e-07 (base)
- `r2` setting `g2/E` to zero: 5.69e-07 (~50% increase from base) <- this is unexpected and seems to be from an increase in internal power and has 0.2 activity on the `CK` pin
- `r2` setting the `r2/CK` port to zero: 1.49e-07 (~60% decrease from base) <- this would be the expected result and has 0.01 activity on the `CK` pin (I would expect that should have been 0.00)

Repeating this analysis for the top clock gate `g1`, results in a similar increase in the power for both registers, despite there being no clocks toggling in the design.
When setting all clock ports to zero `set_case_analysis 0 r1/CK` and `set_case_analysis 0 r2/CK`, the power drops for both registers to 5.44e-08 (basically just the leakage power), which is as expected (where the `CK` pins on each register is 0.00).

Please let me know if you need anything else from the example I've attached or if there was a command I missed to enable the clock gates.

jjcherry56, 2020-11-17:
The underlying issue was pushing constants through statetables. I worked around that by using the gated clock flags to get it to work.

jjcherry56, 2020-11-30:
7662c124 gated clock power

closed 2020-11-30 by jjcherry56