The-OpenROAD-Project/OpenSTA #77: System hangs during full-chip STA with OpenSTA with 15K cells

dineshannayya, 2021-07-07:
I have hierarchical design with 52K cells, When I am trying to do full chip STA with partial selected 15K design + Block Spef.  
I see at timing report command "report_tns",  tool takes more than 30g virtual memory + 100% CPU and hang the system.
My System has: 32MB DDR4 Memory + Intel 10th version i3 with 4 core + 4 thread

I have give -threads 2, still system hangs .. Any suggestion on how to manages this ?  
Is share the our database help ?
What is the best approach for top-level STA.

jjcherry56, 2021-07-07:
The only conceivable way to understand your issue is with a complete test case that runs the design.
I suggest you try running with 1 thread in case it is an issue with multi-threading.

dineshannayya, 2021-07-07:
[exp2.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/6779054/exp2.tar.gz)
Attached the complete test case, with scripts, sdc, verilog, spef,lib

include/lib/*       ----- Library files
include/sdc/*     ---  sdc file
include/netlist/*  --- Netlist
include/spef/*   --- Spef file

sta script:  sta.tcl
run command:  run_cmd

I have tested with latest git files.

Let me know any additional details need to reproduce the issue.

jjcherry56, 2021-07-07:
It is getting lost in what it does not know is power/ground nets used by fillers and pad cells.
It could handle it better, but...
I am going to guess you wrote the verilog from openroad. If so I suggest you NOT use the -include_pwr_gnd option
because the opensta verilog reader does not really respect pg_pins. I would also remove the filler cells with the -remove_cells option.

jjcherry56, 2021-07-08:
There is no liberty definition for the filler cell so it makes black boxes for them with bidirect port, which is a hairball of connections to vdd/vss analogs. Either add a liberty definition for the filler or use set link_make_black_boxes 0 to prevent them from being created.

closed 2021-07-08 by jjcherry56

dineshannayya, 2021-07-08:
I see -remove_cells works for removing sky130_fd_sc_hd__decap* & sky130_fd_sc_hd__diode_* cells, but it does not detect sky130_fd_sc_hd__fill* & ky130_fd_sc_hd__tapvpwrvgnd* 
<pre>% get_lib_cells sky130_fd_sc_hd__decap*
_e07f1d34bc550000_p_LibertyCell _00841d34bc550000_p_LibertyCell _008b1d34bc550000_p_LibertyCell _90921d34bc550000_p_LibertyCell _209a1d34bc550000_p_LibertyCell _00f52535bc550000_p_LibertyCell _b0fb2535bc550000_p_LibertyCell _40032635bc550000_p_LibertyCell _d00a2635bc550000_p_LibertyCell _60122635bc550000_p_LibertyCell
% get_lib_cells sky130_fd_sc_hd__fill*
Warning: cell &apos;sky130_fd_sc_hd__fill*&apos; not found.
% get_lib_cells sky130_fd_sc_hd__diode_*
_30e92934bc550000_p_LibertyCell _10573235bc550000_p_LibertyCell
% get_lib_cells sky130_fd_sc_hd__tapvpwrvgnd*
Warning: cell &apos;sky130_fd_sc_hd__tapvpwrvgnd*&apos; not found.
</pre>

Is it due to liberty definition missing for fill and tap cells ?  what is option remove this cell?
I see link_make_black_boxes 0 not helping, with this command linking of the main module itself not working

jjcherry56, 2021-07-08:
get_**lib**_cells - no liberty, it can't find it.
this is a problem with the sky130 libraries

dineshannayya, 2021-07-09:
Is -remove_cell option can delete only lib_cells ?,  How to delete non-lib module?

jjcherry56, 2021-07-09:
you need to either add a liberty definition or delete the fillers with a tcl script.

jjcherry56, 2021-07-09:
I don't really see why you are using verilog. Why don't you just do the timing in openroad with the database?

dineshannayya, 2021-07-10:
I don't see a way to do full chip **flat** STA with Multiple Hard-macro in Openlane flow. 
I am using standard method of doing top-level STA with reading Macro verilog + SPEF.