The-OpenROAD-Project/OpenSTA #78: Tool not able to parse the net with $  and report as syntax error, unexpected $undefined

dineshannayya, 2021-07-25:
Hi

I see tool not able to parse the .spef file with $ and report as syntax error

<pre>OpenSTA 2.3.0 f2e042ecd8 Copyright (c) 2019, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 &lt;http://gnu.org/licenses/gpl.html&gt;

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying&apos; for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty&apos;.
Warning: input/spef/sdrc_top.spef line 6681, syntax error, unexpected $undefined.
</pre>

Attached the test case and run command

dineshannayya, 2021-07-25:
[Uploading exp3.tar.gz…]()
Attached the Example and run command.
Tool not able to parse net with $ as shows below
*6690 u_wb2sdrc.u_cmdfifo.mem\[0\]\[0\]
*6691 u_wb2sdrc.u_cmdfifo.mem\[0\]\[10\]
*6692 u_wb2sdrc.u_cmdfifo.mem\[0\]\[11\]
*6693 u_wb2sdrc.u_cmdfifo.mem\[0\]\[12\]
*6694 u_wb2sdrc.u_cmdfifo.mem\[0\]\[13\]
*6695 u_wb2sdrc.u_cmdfifo.mem\[0\]\[14\]
*6696 u_wb2sdrc.u_cmdfifo.mem\[0\]\[15\]
*6697 u_wb2sdrc.u_cmdfifo.mem\[0\]\[16\]
*6698 u_wb2sdrc.u_cmdfifo.mem\[0\]\[17\]
*6699 u_wb2sdrc.u_cmdfifo.mem\[0\]\[18\]
*6700 u_wb2sdrc.u_cmdfifo.mem\[0\]\[19\]
*6701 u_wb2sdrc.u_cmdfifo.mem\[0\]\[1\]
*6702 u_wb2sdrc.u_cmdfifo.mem\[0\]\[20\]
*6703 u_wb2sdrc.u_cmdfifo.mem\[0\]\[21\]
*6704 u_wb2sdrc.u_cmdfifo.mem\[0\]\[22\]

dineshannayya, 2021-07-25:
[exp3.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/6874067/exp3.tar.gz)
Updated the Example

jjcherry56, 2021-07-31:
It is illegal spef.

11.3.2.1 Special characters
Any character other than alphanumerics and underscore (_) shall be escaped when used in an identifier in a SPEF file.

closed 2021-07-31 by jjcherry56