The-OpenROAD-Project/OpenSTA #85: Unmatched results when using `disconnect_pin` and `connect_pin` command.

KKA92, 2021-09-02:
When I used your OpenSTA library with the standard TCL commands , and I ran into a weird problem.
My task involved using  fan-in swap a particular cell with another type and calculating the slack improvement.
I tried to use the function disconnect_pin and connect_pin from the STA API to do that.
However, when I tried to revert back to the original fan-in for further processing, the slack values before and after were not matched.

In order to reproduce the problem, please refer to the following steps:

1. unzip gcdfile.zip in your STA examples folder
2. move from examples/gcdfile/sta.tcl to /examples/sta.tcl
3. modify directory in examples/gcdfile/gcd/runs/test2/config.tcl file GCD_ROOT at line 1
4. run with your sta file use command : ../app/sta sta.tcl

`report_checks` result in the original design.
[image]

`report_checks` result after reverting
[image]

this is a link to download my design and sky130 lib files, because this file is too big so I uploaded file to google drive
https://drive.google.com/file/d/16TFSIY1JPvaup6r6TIbA9eYz4SSLaT6C/view?usp=sharing

thanks.

rovinski, 2021-09-02:
This just seems like floating point error, likely from changing the order of operations. The net is still within 2E-7 accuracy. If your time unit is nanoseconds, the difference is less than 1 femtosecond. Plus, the resulting path slack is exactly the same down to the last digit.

I'll leave this to @jjcherry56 to decide if this is buggy, but my bet is that this is expected floating point error, and it's also why we never go more than 3 or so decimal places in the timing report. If you are doing some kind of comparison based on slacks, you need to use rounded comparisons rather than exact comparisons (e.g. [see here](https://floating-point-gui.de/errors/comparison/))

KKA92, 2021-09-03:
Thank you for your answer, @rovinski

I need one more of your confirmation about comparing delay numbers.  In OpenSTA/util/Fuzzy.cc, there are some functions for comparison between two delay values (such as `delayEqual` or `fuzzyEqual`). I wonder can I use these functions as an alternative way for this tutorial: https://floating-point-gui.de/errors/comparison/.

closed 2021-09-05 by jjcherry56

rovinski, 2021-09-06:
@KKA92 that's probably a good idea if that's what you need.