The-OpenROAD-Project/OpenSTA #89: syntax error, unexpected REG

mdzakirhussain, 2022-01-18:
Why is that, sta reports this error "syntax error, unexpected REG".

I have synthesized using yosys, and the netlist file when read in sta, I am facing the aforesaid issue.

jjcherry56, 2022-01-18:
Please provide a testcase that reproduces the issue.

mdzakirhussain, 2022-01-18:
[alu.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893036/alu.txt)
[fa.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893040/fa.txt)
[rca.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893041/rca.txt)
[tb.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893043/tb.txt)

mdzakirhussain, 2022-01-18:
top module is alu.txt

jjcherry56, 2022-01-18:
That is an RTL verilog. OpenSTA can only read structural verilog (post synthesis).

mdzakirhussain, 2022-01-18:
i just provided the files

jjcherry56, 2022-01-18:
For future reference, a text file containing verilog is not a testcase. There is no library data or tcl command file.

mdzakirhussain, 2022-01-18:
[synth.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893054/synth.txt)

mdzakirhussain, 2022-01-18:
This is the file, after synthesis which I am using in opensta. Verilog files , .v is not getting uploaded so I uploaded txt files

mdzakirhussain, 2022-01-18:
I am sorry for my ignorance. Please help me.

mdzakirhussain, 2022-01-18:
[sky130_fd_sc_hd__tt_025C_1v80.txt](https://github.com/The-OpenROAD-Project/OpenSTA/files/7893073/sky130_fd_sc_hd__tt_025C_1v80.txt)

I am using sky130_fd_sc_hd__tt_025C_1v80.lib

jjcherry56, 2022-01-19:
synth.v/txt is not structural verilog. it has reg and always statements in it.

closed 2022-01-19 by jjcherry56