The-OpenROAD-Project/OpenSTA #95: `report_power` has unknown units, `set_cmd_units` is unreliable

kop316, 2022-04-03:
Hello!

I am trying to debug what units OpenSTA uses in OpenROAD for `report_power`. It is not in the documentation. 

I attempted to experiement with `set_cmd_units` here: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/sta.tcl#L51

and I added `report units` here: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/sta.tcl#L135

Some experiements I did
```
set_cmd_units -time ns -capacitance pF -current A -voltage V -resistance kOhm -distance um
```
Yields:

```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1A
 power 1nW
 distance 1um
unit_report_end
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             8.41e-05   1.77e-05   1.52e-10   1.02e-04  44.3%
Combinational          6.74e-05   6.06e-05   1.17e-09   1.28e-04  55.7%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  1.51e-04   7.83e-05   1.32e-09   2.30e-04 100.0%
                          65.9%      34.1%       0.0%
power_report_end
area_report
```

```
set_cmd_units -time ns -capacitance pF -current mA -voltage V -resistance kOhm -distance um
```

yields
```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
unit_report_end
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             8.41e-05   1.77e-05   1.52e-10   1.02e-04  44.3%
Combinational          6.74e-05   6.06e-05   1.17e-09   1.28e-04  55.7%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  1.51e-04   7.83e-05   1.32e-09   2.30e-04 100.0%
                          65.9%      34.1%       0.0%
power_report_end
area_report
```

Finally,
```
set_cmd_units -time ns -capacitance pF -current A -voltage V -resistance kOhm -distance um -power mW
```
yields
```
 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1A
 power 1nW
 distance 1um
unit_report_end
power_report

===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power
----------------------------------------------------------------
Sequential             8.41e-05   1.77e-05   1.52e-10   1.02e-04  44.3%
Combinational          6.74e-05   6.06e-05   1.17e-09   1.28e-04  55.7%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  1.51e-04   7.83e-05   1.32e-09   2.30e-04 100.0%
                          65.9%      34.1%       0.0%
power_report_end
area_report
```

I would have expected modifying any of those (espeically `-power`) to modify what `report_power` reports, but as far as I can tell, nothing I do influences the report_power units.

kop316, 2022-04-04:
Looking through your source:

https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/tcl/Sdc.tcl#L284

It looks like you should have:

```
set_unit_values "power" -distance "w" keys
```

in the `set_cmd_units` proc, but it doesn't exist.

jjcherry56, 2022-04-04:
It is documented in the OpenSTA doc found in src/sta/doc/OpenSTA.pdf.
"Power values are reported in watts."

closed 2022-04-04 by jjcherry56