The-OpenROAD-Project/OpenSTA #96: set_cmd_units does not set power

kop316, 2022-04-04:
Looking through your source:

https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/tcl/Sdc.tcl#L284

It looks like you should have:

set_unit_values "power" -distance "w" keys

in the set_cmd_units proc, but it doesn't exist.

Sorry, I added this in the other issue, but I should probably make it its own issue.

jjcherry56, 2022-04-04:
correct. however I am not sure anything actually uses the power units for commands. report_power uses watts because that made the most sense when exponentials are used to make the reporting format line up.

closed 2022-04-04 by jjcherry56

kop316, 2022-04-04:
The Default OpenLane `sta.tcl` script modifies the units in `report_units` to be `nW` vs w, see:

https://github.com/The-OpenROAD-Project/OpenLane/issues/1032#issuecomment-1087081961

if you modify `sta.tcl` to only have `report_units`, it reports it as:

```
 time 1s
 capacitance 1F
 resistance 1ohm
 voltage 1v
 current 1A
 power 1W
 distance 1m
 ```
 but, `sta.tcl` shows it as:

```

 time 1ns
 capacitance 1pF
 resistance 1kohm
 voltage 1v
 current 1mA
 power 1nW
 distance 1um
```
So something is happening to modify the power units and it oes not get changed back