The-OpenROAD-Project/OpenSTA #98: Does OpenSTA support PBA in addition to GBA?

tdene, 2022-04-14:
Path-Based Analysis, while much more computationally difficult, can lead to significant pessimism reductions when compared to the more computationally-simple Graph-Based Analysis.

In my experience, layouts are commonly timed using GBA during the initial design steps, before switching to PBA near the end of the design process in order to fully optimize the design.

Does OpenSTA support PBA?

If not, are there current plans to add support for PBA, and are there ways in which outside contributors may be able to assist in the implementation of PBA?

QuantamHD, 2022-04-14:
@maliberty Can you answer this question. It's for Google.

tspyrou, 2022-04-14:
@tdene and @QuantamHD at this time OpenSTA does not support PBA. As you mentioned it is an important feature to reduce pessimism in STA and is useful in both the construction analysis and signoff analysis. This is something that we would like to add along with CCS and SI support.

closed 2024-08-12 by maliberty