The-OpenROAD-Project/OpenSTA #99: Signal Integrity support

Blebowski, 2022-06-02:
Hello,

we are looking at OpenSTA in our ASIC design team, and we are starting an evaluation on our design.
We would like to compare its output with PrimeTime. Feature-wise, I have noticed that Signal Integrity
is not yet supported by OpenSTA.

Is there an interest/intent to implement Signal integrity support (negative impact on WNS, or reporting of noise which
could potentially flip logic level)? If so, is there something where we could potentially start if we found the
time to dedicate an effort to this?

Blebowski, 2022-06-02:
Thanks for such a quick reply. What existing signal integrity solution would you like to integrate? Sparselizard?

jjcherry56, 2022-06-02:
The signal integrity code was developed by a parallax customer years ago. Sparselizard is just a sparse matrix solution, which is light years away from a signal integrity support.

Blebowski, 2022-06-02:
Ok, thanks for explaining. I only knew that Kicad is trying to use sparselizard for some basic SI/PI PCB stuff. As for me, I am really a SI noob (at least when it comes to practical EDA SW development, I have just some rudimentary SI + FEM basics from EE classes). Just out of curiosity, what is the approach to do SI in STA? I guess field solver is not really needed, since geometry of the circuit is not present, but parasitics are already there from SPEF. Is it just lumped element model of capacitance on each wire, and the tool increases the delay and transitions based on added capacitance?

Blebowski, 2022-06-02:
Or does the tools like PrimeTime account for actual logic values on adjacent wires (IMHO they can't they dont simulation functionality of the circuit)? I mean, lets have two adjacent wires (agressor(A)/victim(V)) with mutual cap C. If A has edge transition T, then amount of voltage induced on V can be calculated from: drive strength of A, T and C ? Or do the tools account for logic level difference between A and V at the moment when edges are being propagated through A/V ? Or does it always consider the worst case ?

mithro, 2022-06-04:
FYI - @QuantamHD

Ceridli, 2022-07-17:
Signal integrity issue for VLSI is somewhat different.  Its effect depends on the arrival times of the signals to have a significant
effect on timing.  That depends on a lot of things. It is better to view it as a random effect on the load.

Blebowski, 2022-07-18:
@Ceridli, could you maybe send me some scientific articles / technical materials about this? I would like to understand how it works in STA engines.

Ceridli, 2022-07-18:
Unfortunately, I stopped writing for quite some time. However, it is not rocket science. There are two issues here: When there is capacitive coupling between the nets, (i) from driven net there could capacitive current injection to nets which are peacefully sitting there.  (2) if there are busses with similar drivers running in parallel, the effective load changes. In the first case, the change in voltage at the victim is Cc/(Cc+ Cv)*VDD, ignoring that the victim has a driver of its own. Here Cc is coupling  capacitance, and Cv is lumped capacitance of the victim. If this exceeds the threshold,  a secondary propagation may be started .  So, one needs find these two values, and make sure that thresholds are not crossed. The second problem can be addressed by offset buffer placement to avoid  simultaneous switching. Its effectiveness depends on ratio of buffer delay to the total bus delay.  In my opinion, it is better to treat this as a potentially local /dynamic variation of load.

Ceridli, 2022-07-18:
By the way, for the first effect, it is just a static calculation.  You don't necessarily need an STA engine. You can run through actual
simulation, with passive victim drivers affected by their agressive neigbor. These are linear circuits, one can go through them
using Spice in no time.

maliberty, 2024-08-12:
Issues or PRs should be filed with https://github.com/parallaxsw/OpenSTA if still relevant.  This is effectively a fork (though not strictly for historical reasons).

closed 2024-08-12 by maliberty

themperek, 2024-08-13:
There seems some support for the impact on timing but I cannot see support for `reporting of noise`.