The-OpenROAD-Project/OpenSTA #104: set_clock_uncertainty -rise -fall

Chopper455, 2022-07-27:
Tcl procedure of SDC command set_clock_uncertainty doesn't account -rise and -fall flags for inter-clock constraint (-from and -to).
It only checks if they were illegally used for single-target constraint.
This causes uncertainty to stick to wrong points.

jjcherry56, 2022-07-30:
-rise_to and -fall_to are an equivalent way to specify the edge that are supported, but I added support for -rise/-fall because they are in the sdc spec despite being redundant

closed 2022-07-30 by jjcherry56

Chopper455, 2022-08-01:
The point is that these flags aren't processed even if user specifies them together with -to arguments.

>   if { [info exists keys(-to)] } {
>     set to_key "-to"
>     set to_rf "rise_fall"
>   } elseif { [info exists keys(-rise_to)] } {
>     set to_key "-rise_to"
>     set to_rf "rise"
>   } elseif { [info exists keys(-fall_to)] } {
>     set to_key "-fall_to"
>     set to_rf "fall"
>   } else {
>     set to_key "none"
>   }

jjcherry56, 2022-08-01:
look at 1aa348f set_clock_uncertainty -rise -fall parsing

Chopper455, 2022-08-02:
Sorry to have bothered you. Thank you.