The-OpenROAD-Project/OpenSTA #106: How to use opensta as timing engine through network adaptor

taoli4rs, 2022-08-13:
Hi Experts,
While browsing the wiki documents, I'm interested by "using OpenSTA as timing engine through network adapter". Several questions:

1. How to create a network with this adapter? Is there an example/demo code?
2. I assume this network is rendered with timing/power data, which means once this network is built up successfully it is the only input data (no more lib, sdf, netlist requried) for OpenSTA for timing check and report. Is that true?

Thanks a lot!

jjcherry56, 2022-08-13:
see doc/StaApi.txt
ConcreteNetwork.hh is an example network adapter for OpenSTA to run stand alone

taoli4rs, 2022-08-16:
Thank you JJ. If I understand it correctly, the standard flow to build a concrete network including reading liberty file, sdf file and verilog netlist. Which means that a liberty file to define cell internal delay and a sdf to define interconnect delay is a must to create it for timing analysis to be carried up by opensta.

jjcherry56, 2022-08-16:
Not really. The whole point of using a network adapter is to NOT copy an external netlist/network. I think you are confusing many different concepts in your reply. The only point of using a network adapter is to use opensta with existing external network/netlist data structures. Verilog is irrelevant. The netlist could come from DEF. The ConcreteNetwork exists for stand alone opensta when it is NOT interfaced to an external tool. SDF is used to override delay calculation and is NOT required. SDF is not just for interconnect delays, it can also annotate gate delays.

taoli4rs, 2022-08-16:
Thanks JJ. I'm trying to educating myself to understand the ConcretNetwork adapter by using example1.tcl, obviously in that case the read lib, verilog and sdf to create concrete network mis-led me from what you just described. 

Then back to my first question: To adopt OpenSTA as timing engine in a software tool, what is the best method to do so? Any suggestion on minimum input data (for OpenSTA module) and what the major class/module (ConcreteNetwork, timing data base, timing calculator?) in OpenSTA I should focus on? 

Thank you.

akashlevy, 2024-08-12:
For anyone attempting to use OpenSTA as a timing engine, I would recommend looking at how OpenROAD does this as a reference.

closed 2024-08-12 by maliberty