The-OpenROAD-Project/OpenSTA #108: OpenSTA unnecessarily ignores output pin activity factor annotations

taylor-bsg, 2022-08-16:
https://github.com/The-OpenROAD-Project/OpenSTA/blob/12022edc37acf0370f3686eb4ce05c3490b4a7ce/tcl/Power.tcl#L244

jjcherry56, 2022-08-17:
Perhaps, but I'm struggling to understand why it matters. It can't propagate anywhere from an output and I don't think it is used because it is not an instance pin.

taylor-bsg, 2022-08-17:
Thanks for your reply!

I am generating set_power_activity commands using a SAIF-> OpenSTA set_power_activity converter that I extended (see https://github.com/bsg-external/cppSaif/blob/master/src/set_power_activity.cpp).

It successfully annotates duty and activity factors for the inputs of the gates; but it silently ignores the attempts to annotate the outputs. 

I found just by deleting this one line of code I get vastly more accurate power numbers, since it is using real data for output pins instead of propagated values.

jjcherry56, 2022-08-17:
Ya, I was thinking of output PORTS, not PINs.
I pushed the change.

closed 2022-08-17 by jjcherry56