The-OpenROAD-Project/OpenSTA #110: setting power activity on register overwrites activity on downstream logic and prevents it from being changed

taylor-bsg, 2022-08-16:
```
%  source example2.tcl
Startpoint: r2 (rising edge-triggered flip-flop clocked by clk)
Endpoint: r3 (rising edge-triggered flip-flop clocked by clk)
Path Group: clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock clk (rise edge)
   0.00    0.00   clock network delay (ideal)
   0.00    0.00 ^ r2/CK (DFF_X1)
   0.23    0.23 v r2/Q (DFF_X1)
   0.08    0.32 v u1/Z (BUF_X1)
   0.10    0.42 v u2/ZN (AND2_X1)
   0.00    0.42 v r3/D (DFF_X1)
           0.42   data arrival time

  10.00   10.00   clock clk (rise edge)
   0.00   10.00   clock network delay (ideal)
   0.00   10.00   clock reconvergence pessimism
          10.00 ^ r3/CK (DFF_X1)
  -0.16    9.84   library setup time
           9.84   data required time
---------------------------------------------------------
           9.84   data required time
          -0.42   data arrival time
---------------------------------------------------------
           9.43   slack (MET)

%  puts [get_property -object_type pin u1/A activity]
1.00000e+07 0.500 input
% set_power_activity -pins u1/A -activity 1 -duty 0.15
%  puts [get_property -object_type pin u1/A activity]
1.00000e+07 0.500 input
% report_power -instances u1               
   Internal  Switching    Leakage      Total
      Power      Power      Power      Power
--------------------------------------------
   1.63e-07   5.64e-08   1.56e-08   2.35e-07 u1
%  puts [get_property -object_type pin u1/A activity]
1.00000e+08 0.150 user
%  set_power_activity -pins r2/D -activity 0.25 -duty 0.10
% report_power -instances u1   
   Internal  Switching    Leakage      Total
      Power      Power      Power      Power
--------------------------------------------
   1.63e-08   5.64e-09   1.56e-08   3.75e-08 u1
%  puts [get_property -object_type pin u1/A activity]
1.00000e+07 0.500 input
% set_power_activity -pins u1/A -activity 1 -duty 0.15
%  report_power -instances u1  
   Internal  Switching    Leakage      Total
      Power      Power      Power      Power
--------------------------------------------
   1.63e-08   5.64e-09   1.56e-08   3.75e-08 u1
%  puts [get_property -object_type pin u1/A activity]
1.00000e+07 0.500 input
% 
```

taylor-bsg, 2022-08-16:
@jjcherry56 Thanks for looking at these -- OpenSTA is a lot of fun to use!

taylor-bsg, 2022-08-17:
@jjcherry56 this is the biggest blocker of all the things I reported .. I don't know how to fix it.

jjcherry56, 2022-08-18:
fixed by 
cbf6370 power activity propagtion from annootated reg pins

closed 2022-08-18 by jjcherry56