The-OpenROAD-Project/OpenSTA #113: Verilog assign and SPEF annotation

shalan, 2022-08-18:
Does OpenSTA handle assign statements in the netlist while loading SPEF files? 

Loading a SPEF file to annotate a netlist that has assign statements emits warnings about nets being not connected. For example: 

```
Warning: ~/spef/caravel-nom-t.spef line 6060, *2803 not connected to net gpio_clock_1_shifted\[0\].
```

The net represented by `*2803` is connected to  `gpio_clock_1_shifted\[0\]` using an `assign` statement in the netlist.

maliberty, 2022-08-18:
You should provide a test case

jjcherry56, 2022-08-21:
Short of providing a testcase can you at least describe what nets the assign connects? Input? Outputs? Hierarchical module ports? The probability of fixing the problem you have is much smaller if I have to guess what the netlist looks like.

tspyrou, 2022-09-26:
@shalan is this a hierarchical spef flow? If not it would be good to get the testcase so it can be fixed.

jjcherry56, 2022-09-30:
I tried to reproduce this with the attached verilog/spef but it worked as expected.
Maybe you can modify the attached test case to reproduce your issue.

[sta113.zip](https://github.com/The-OpenROAD-Project/OpenSTA/files/9686972/sta113.zip)

closed 2022-12-26 by jjcherry56