The-OpenROAD-Project/OpenSTA #116: hierarchical top level macro-to-marco spef annotation issue

kareefardi, 2022-08-24:
This is a test case that highlights an issue with top level annotation. The test case is a design with two macros. The macros are identical and each consists of two inverters with enough distance between them to cause a delay. The macros are connected through a long net with high resistance and capacitance at the top level. However, the sta report doesn't show any capacitance or extra delay on the top level net in between macros. The test case has all the needed files (.lib,.verilog,.spef) and a runner `sta.bash` script that runs the combinations of the presence and absence of top level spef and macros spef for comparison
[test-case.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/9417543/test-case.tar.gz)

maliberty, 2022-08-24:
-increment doesn't appear to stitch parasitics so much as it silently ignores any reannotation of a net (https://github.com/The-OpenROAD-Project/OpenSTA/blob/bb50746f72ab112819570abce352684f9114f624/parasitics/SpefReader.cc#L450)

There should be a message.  @tspyrou comments that the expected stitching behavior might not have been implemented but will have to wait for @jjcherry56 to resolve.

kareefardi, 2022-08-24:
Thanks. It' also worth noting that we still don't get delays at the top level inner net even with only loading the top level spef.

maliberty, 2022-08-24:
That is a problem as well.  Since solving it wouldn't really be a solution to your overall need I'll leave it for Cherry to look into when he has time.  If you feel differently let me know.

jjcherry56, 2022-08-25:
There is no support for hierarchical spef annotation.

kareefardi, 2022-08-26:
@maliberty so the recommended mythology won't work?
Also in that case what is supported with spef annotation?

maliberty, 2022-08-29:
It is correct that this issue breaks the methodology I recommended.

kareefardi, 2022-09-04:
Is there a plan to add this feature?

jjcherry56, 2022-10-08:
OpenSTA commit 0722453 adds support for hierarchical spef annotation.

closed 2022-10-08 by jjcherry56