The-OpenROAD-Project/OpenSTA #117: specifying clock rise/fall slopes for propagated clocks

taylor-bsg, 2022-08-25:
Is this possible? I checked the docs and did not see any mention of it.

Thanks!

jjcherry56, 2022-08-25:
Not really. You want the delay calculation for gate delays in the clock network but not the slews? It is a strange request. Seem like you should be using ideal clocks with set_clock_latency and set_clock_transition.

taylor-bsg, 2022-08-26:
Thanks for your reply. What I mean is, let's say you have a clock tree and you have parasitics for it. The clock tree still has a source, for which we want to specify a slew; what is the mechanism for specifying this?

jjcherry56, 2022-08-26:
set_input_transition

closed 2022-08-26 by jjcherry56

taylor-bsg, 2022-10-11:
Actually I am saying, I want it to time the clock tree and everything with
slews etc, but the clock input pin itself has a slew.

taylor-bsg, 2022-10-11:
Okay great, I did not know this could be applied to a clock signal as well.

M