The-OpenROAD-Project/OpenSTA #122: power report

sfmth, 2022-10-03:
I ran my RTL through Openlane and there is a file in reports that looks like this:
```
===========================================================================
 report_power
============================================================================
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.21e-03   2.66e-04   1.73e-08   1.48e-03  33.8%
Combinational          1.55e-03   1.35e-03   1.16e-07   2.90e-03  66.2%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.76e-03   1.62e-03   1.33e-07   4.38e-03 100.0%
                          63.0%      37.0%       0.0%
```
I was wondering where these values come from and whether it is possible to get an estimation of energy consumption for a specific operation in my design.

rovinski, 2022-10-03:
You can see more info about the power reports in the [OpenSTA documentation](https://github.com/The-OpenROAD-Project/OpenSTA/tree/master/doc). In the code, it is located in `liberty/*Power.cc`. Generally, default activity factors are assigned to inputs and the activity is propagated through the design to calculate the power. You can use `set_power_activity` to set the activity factor for specific inputs.

It sounds like you are looking for vector-based power calculation (providing a time-sequence of input data) which is not currently supported. However, I understand that it is a planned feature.

sfmth, 2022-10-16:
How can I use `set_power_activity` when running a design through openlane or after it?

sfmth, 2022-10-16:
and what are the default activity and duty values?

jjcherry56, 2022-12-26:
OpenSTA now supports reading a VCD file for activities. See the read_power_activities command.
See the set_power_activity command doc for default input activities.

closed 2022-12-26 by jjcherry56