The-OpenROAD-Project/OpenSTA #128: create_generated_clock

Fraserbc, 2022-11-06:
Hi,

I have two issues with `create_generated_clock`. The first is when trying to use the `master_clock`  argument without providing `source` is errors out with `missing -source argument.`, but according to [this](https://www.intel.com/content/www/us/en/docs/programmable/683432/22-1/tcl_pkg_sdc_ver_1-5_cmd_create_generated_clock.html) it should work.

My second issue is that it requires `source` to be a pin or a port as it errors with `master_pin type 'Net' is not a pin or port.`, not allowing nets to be used.

vijayank88, 2022-11-07:
@Fraserbc 
Refer following doc: https://github.com/The-OpenROAD-Project/OpenSTA/blob/master/doc/OpenSTA.pdf

Fraserbc, 2022-11-07:
I've just read through it again and can't see anything showing how to create a generated clock using a net as a source. Is there any workaround?

vijayank88, 2022-11-07:
@Fraserbc 
Can you attach test case with `.sdc` file you're facing the issue?

jjcherry56, 2022-11-07:
Nets are not supported for generated clock sources. It is actually pretty strange and counter to the documentation (that I wrote) that nets are supported for the clock root. The net is converted to the driver pin(s) of the net. The SDC to do the equivalent is
[get_pins -of_object [get_net CLK_DIV_2] -filter "direction == output"]

Fraserbc, 2022-11-07:
@jjcherry56 Thanks so much! That's exactly what I was looking for. By any chance do you have any good places to learn about SDC constraints, specifically things like that `[get_pins -of_object [get_net CLK_DIV_2] -filter "direction == output"]`?

closed 2022-12-26 by jjcherry56