The-OpenROAD-Project/OpenSTA #129: Activity propagation for trigger

stevvik, 2022-11-22:
Hello!

We faced with the issue during activity propagation through trigger pins.
We assume, that in case of frequency on trigger data input pin (D) is greater than on clock pin (CK), the frequency on output pins (Q or QN) must be half of clock pin frequency (Fq=Fck/2). The same must be applicable for activity propagation: activity on output pin must be twice lower than on clock input pin.
But in OpenSTA we see that output pins activities are equal to data input activity!

Simple testcase with single trigger and OpenSTA commands we used is attached.
Top level runfile is  "run_opensta".
[testcase.tgz](https://github.com/The-OpenROAD-Project/OpenSTA/files/10067460/testcase.tgz)

jjcherry56, 2022-12-02:
You are correct that the D->Q activity propagation should not be greater than the active edge count.
Commit fdba9eb correct this.

I should point out that your testcase is missing a clock definition, which is how the STA knows where the clocks are (NOT the clock pin on the liberty cell).

Also, the units in your set_power_activity commands are incorrect. As the doc explains they are in **transitions per clock cycle**, not transitions per second.

closed 2022-12-02 by jjcherry56

stevvik, 2022-12-02:
@jjcherry56 ,
I decided to draw you a picture to simplify my explanations (I can send you testcase later if you need):

[image]

stevvik, 2022-12-02:
[image]

jjcherry56, 2023-01-17:
That is a clock network which has to be defined using create_generated_clock.