The-OpenROAD-Project/OpenSTA #130: OpenSTA creates Verilog with implicit wires

antonblanchard, 2022-12-05:
OpenSTA is creating implicit wires on unconnected buses. An example is attached, which creates the following Verilog:

```
module module1 ();

 module2 m (.A0({_NC1,
    _NC2}));
endmodule
```

It's a minor issue, a result of the gate level test using `default_nettype none` somewhere and fixed by adding `default_nettype wire`. I did notice OpenSTA emits wire statements elsewhere, so hopefully it could do the same for this case.

Testcase: [sta-implicit-wire.tar.gz](https://github.com/The-OpenROAD-Project/OpenSTA/files/10151646/sta-implicit-wire.tar.gz)

jjcherry56, 2022-12-05:
ya, a bit tricky because those net names aren't known until deep into the write...

jjcherry56, 2022-12-17:
fixed in 46c6838

closed 2022-12-17 by jjcherry56