The-OpenROAD-Project/OpenSTA #131: documentation improvement request for -min and -max

mattvenn, 2022-12-24:
in set_input_delay and set_output_delay command documentation:

-max This is the minimum arrival time.
-min This is the maximum arrival time.

and from quartus docs:

-max | Applies value as maximum data required time
-min | Applies value as minimum data required time

It's confusing for me as a beginner.

also would be good to:
 
* state that it's in ns 
* that it's relative to before the edge 
* so a negative number would move it after the edge

rovinski, 2022-12-25:
> state that it's in ns

It actually is not in ns, it is in the time unit specified by your liberty `.lib` file. sky130 for example is ns, but ASAP7 is ps.

jjcherry56, 2022-12-26:
The min/max doc is definitely backwards.
ALL commands use the liberty units. I added this to the opening section.
The set_output_delay doc was pretty twisted so I rewrote it. Hopefully it is clearer now.

closed 2022-12-26 by jjcherry56

mattvenn, 2022-12-27:
thanks for the fast fix on this - appreciated!

mattvenn, 2022-12-27:
Couple of questions - where is the best place to ask for support for OpenSTA?
I'm trying to learn to read the reports - is there an annotated report somewhere I can study?