The-OpenROAD-Project/OpenSTA #132: Memory leak while reading liberty

spacemonkeydelivers, 2022-12-29:
Hi,

I've been using OpenSTA and at some point I've built my project with ASAN and it reported a memory leak in OpenSTA:
```
==58452==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 6640 byte(s) in 166 object(s) allocated from:
    #0 0x563c8938736d in operator new(unsigned long) (...)/llvm/llvm-project-15.0.0.src/compiler-rt/lib/asan/asan_new_delete.cpp:95:3
    #1 0x563c896c6410 in sta::LibertyReader::beginTestCell(sta::LibertyGroup*) ((...)/build/app/app+0x6e7410)
```
If you'd take a look into liberty/LibertyReader.cc for **beginTestCell** and **endTestCell** you'd notice that test_cell_ is allocated but is never freed.
Or am I missing something?

Thanks

jjcherry56, 2022-12-29:
LibertyReader::beginTestCell transfers ownership to cell_ and it is freed in LibertyCell::~LibertyCell

closed 2022-12-29 by jjcherry56

jjcherry56, 2022-12-29:
I suspect your library has more than one test cell in a cell, in which case the first one leaks. If that is the case I can update it to delete the first one and warn. I would say the library is broken.

commit 3cf999c3 (2024-12-23): closes #132 spef net missing escapes