The-OpenROAD-Project/OpenSTA #133: Strange power report with a hierarchical design

rbarzic, 2023-01-11:
Hello,

I'm trying to evaluate the power reporting capabilities of OpenSTA on a hierarchical netlist and spef generated by Cadence (Innovus).

After readling the libraries, netlist  and the spef file for the design, I do :
read_power_activities -vcd waves.vcd  -scope tb/U_CHIP/U_CPU
then
report_power -instances { U_CPU/U_RF/FE_COEC10229_n_1608 U_CPU/U_RF }

And I get :
  Internal  Switching    Leakage      Total
      Power      Power      Power      Power (Watts)
--------------------------------------------
   3.32e-15   3.45e-15   0.00e+00   6.77e-15 U_CPU/U_RF/FE_COEC10229_n_1608
   0.00e+00   0.00e+00   0.00e+00   0.00e+00 U_CPU/U_RF

I would expect the report for the U_CPU/U_RF to include at least the number for the instances below its level

(I'm using the master branch, commit 6b8ab8aa)

jjcherry56, 2023-01-11:
That is because the instances arg does not support hierarchical instances (which would be a good enhancement).
The instances arg is mostly used for debugging.

rbarzic, 2023-01-11:
Ok thank you, But what about the command report_power without option? Will this report the power for the top-level without the contribution of the subblocks?

jjcherry56, 2023-01-11:
yes

jjcherry56, 2023-01-20:
commit  4b009e5 report_power -instance hierarchical

closed 2023-01-20 by jjcherry56